florianfesti/boxes #438: Divider tray - Open shapes plus enhancement

leyton01, 2022-10-01:
Generating using the default setting for the divider tray and importing into Lightburn shows some of the dividers not joined in the bottom left corner (open shape).
[image]

Also wondering if having a dividers with half tabs (vertically) rather than half tabs (horizontally) would be a good option.  These could be placed next to each other easily as long as the dividers face the same direction.  See rendered example.
[image]

Thanks for all the good work!

florianfesti, 2022-10-01:
Please try out the new divider type and close this ticket if everything is OK!

florianfesti, 2023-01-05:
Guess I waited enough for feedback...

closed 2023-01-05 by florianfesti