florianfesti/boxes #440: 502 Proxy error

lostbastard, 2022-10-04:
This problem has appeared for two days when i try to generate a dxf file

Proxy Error
The proxy server received an invalid response from an upstream server.
The proxy server could not handle the request [GET /boxes.py/UniversalBox](https://www.festi.info/boxes.py/UniversalBox).

Reason: Error reading from remote server

https://www.festi.info/boxes.py/UniversalBox?language=en

[image]
[image]

nickywittstock, 2022-10-10:
Im also having this issue, but only with the hinged box.

florianfesti, 2022-10-10:
Looks like the server is a bit unreliable. But it seems to work now. Or do you have a link that still fails?

Blancanegra, 2022-10-10:
I'm still getting the same error for UniversalBox, but not for HingeBox.

nickywittstock, 2022-10-10:
https://www.festi.info/boxes.py/UniversalBox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&Hinge_grip_percentage=0&Hinge_outset=0&Hinge_outset=1&Hinge_pinwidth=0.5&Hinge_style=outset&Hinge_axle=2.0&Hinge_grip_length=0&Hinge_hingestrength=1&CabinetHinge_bore=3.2&CabinetHinge_eyes_per_hinge=5&CabinetHinge_hinges=2&CabinetHinge_style=inside&CabinetHinge_eye=1.5&CabinetHinge_play=0.05&CabinetHinge_spacing=2.0&Lid_angle=90.0&Lid_hole_width=0&Lid_second_pin=0&Lid_second_pin=1&Lid_spring=both&Lid_style=rectangular&Lid_surroundingspaces=2.0&Lid_edge_width=1.0&Lid_extra_length=0.0&Lid_finger=3.0&Lid_play=0.05&Lid_space=2.0&Lid_width=1.0&Click_angle=5.0&Click_bottom_radius=0.1&Click_depth=3.0&RoundedTriangleEdge_height=50.0&RoundedTriangleEdge_r_hole=2.0&RoundedTriangleEdge_radius=30.0&RoundedTriangleEdge_outset=1.0&Mounting_d_head=6.5&Mounting_d_shaft=3.0&Mounting_margin=0.125&Mounting_num=2&Mounting_side=back&Mounting_style=straight+edge%2C+within&HandleEdge_height=20.0&HandleEdge_hole_height=75.0&HandleEdge_hole_width=40%3A40&HandleEdge_on_sides=0&HandleEdge_on_sides=1&HandleEdge_radius=10.0&HandleEdge_outset=1.0&Flex_stretch=1.05&Flex_connection=1.0&Flex_distance=0.5&Flex_width=5.0&top_edge=i&bottom_edge=s&x=100.0&y=100.0&h=100.0&outside=0&outside=1&vertical_edges=finger+joints&lid=default+%28none%29&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&render=1

nickywittstock, 2022-10-10:
> I'm still getting the same error for UniversalBox, but not for HingeBox.

same here.

nickywittstock, 2022-10-10:
[image]

nickywittstock, 2022-10-10:
[image]

nickywittstock, 2022-10-10:
I get the above error when trying to create this universal box

myokaski, 2022-10-10:
Works ok from my friend's iPhone and MacBook. Same issue from all other devices and browsers I have

leethomsonart, 2022-10-12:
I am having the same error on Universal Boxes but not any others. I'm using windows/firefox. I tried a new tab, a new window, and a complete reboot.

nickywittstock, 2022-10-13:
This is still an ongoing problem, has anyone been able to solve it? I am getting desperate as we are inching closer to Christmas and I rely on this site for custom made boxes. Please Help

nickywittstock, 2022-10-15:
Call me old fashioned, but I still run windows 7 and love it. I solved the problem on my side by using internet explorer.

Kukinac, 2022-10-19:
Hi, I get the same problem with UBox:

_Proxy Error
The proxy server received an invalid response from an upstream server.
The proxy server could not handle the request [GET /boxes.py/UBox](https://www.festi.info/boxes.py/UBox).

Reason: Error reading from remote server

Apache/2.4.6 (CentOS) Server at www.festi.info Port 443_

But few other boxes, that I tried, seem to work okay.
Pity that I need the UBox :D :D
Has anyone find how to solve it?

nickywittstock, 2022-10-19:
> Hi, I get the same problem with UBox:
>
> _Proxy Error The proxy server received an invalid response from an upstream server. The proxy server could not handle the request [GET [...]

Internet Explorer works. Seems to be an issue when using Chrome

florianfesti, 2023-01-01:
OK, looks like we got crawled to death. Added a `robots.txt` to the two expensive services running on the VM. I hope this fixes the issue.

closed 2023-01-01 by florianfesti

florianfesti, 2023-01-02:
Also moved the static content (images, CSS, ...) to the github.io site which already hosts the documentation.