florianfesti/boxes #441: Instalation problem

lostbastard, 2022-10-05:
Hi

You made a great program. I use it often. I tried to install it according to the instructions you provided on the site.

https://florianfesti.github.io/boxes/html/install/windows.html#native

And I ran into the following problem
[Errno 2] No such file or directory

[image]

I would be grateful for your help and advice. How could I solve this problem.

Ranthalion, 2022-11-29:
Based on your screenshot, you may have 2 nested folders called boxes-master.  You may need to `cd boxes-master` again to get to the correct folder.

florianfesti, 2022-12-31:
Guess the issue has resolved itself one way or another...