florianfesti/boxes #442: pictures HingeBox

YodaMaster18, 2022-10-05:
Dear Forian,

Hereby my pictures of a HingeBox.LxHxD=300x120x200mm outside

[image]
[image]
[image]
[image]

florianfesti, 2022-12-25:
Ehm, the hinges go on the inside to allow the lid to open completely - as shown in the image. I'll add a sentence or two of assembly instructions...

florianfesti, 2022-12-25:
Sorry, I didn't quite get that is is meant to be a bug report. Otherwise I might have answered a bit quicker...

commit 9729ca6d (2022-12-25): HingeBox: Add assembly instructions

Resolves: #442
closed 2022-12-25 by florianfesti