florianfesti/boxes #445: Build / host instructions?

schvenk, 2022-10-11:
This is great! Any chance of some quick instructions (particularly for those of us with limited Python knowledge) for how to self-host the web version and/or build the Inkscape version? Thanks!!

schvenk, 2022-10-11:
Ah! Found the Inkscape instructions. I'd still love to know how to host my own version of the website at https://www.festi.info/boxes.py/ though.

Ranthalion, 2022-11-29:
Do these instructions solve your issue?
https://florianfesti.github.io/boxes/html/install/windows.html#native

schvenk, 2022-12-05:
Yes, that works! I'm on Mac rather than Windows but the WSL instructions work perfectly.

Ranthalion, 2022-12-05:
There are also Mac instructions at https://florianfesti.github.io/boxes/html/install/macos.html
If this has solved the issue, would you mind Closing the issue here in github?

closed 2022-12-05 by schvenk