florianfesti/boxes #448: UniversalBox generates proxy error on https://www.festi.info/boxes.py/

SuksAE, 2022-10-19:
To reproduce open [https://www.festi.info/boxes.py/UniversalBox](https://www.festi.info/boxes.py/UniversalBox)
Don't change any settings, just press "Generate". 
This should open an new tab showing the .svg file, but instead an error message is shown.

Tested with Firefox V106.0 (current version) on Windows 10 (current patches installed).
Same result also on an Android phone with Firefox or Chrome browser. 

Error message:

```
Proxy Error

The proxy server received an invalid response from an upstream server.
The proxy server could not handle the request [GET /boxes.py/UniversalBox](https://www.festi.info/boxes.py/UniversalBox).

Reason: Error reading from remote server

Apache/2.4.6 (CentOS) Server at www.festi.info Port 443
```

SuksAE, 2022-10-19:
Tested with "boxesserver" script on local PC - this works as intended...

florianfesti, 2022-12-31:
Duplicate of #440

closed 2022-12-31 by florianfesti