florianfesti/boxes #452: Bonus chads next to finger joints in inner x trays

nicjansma, 2022-11-12:
**Describe the bug**
Hi!  Just came across this project and it looks amazing.

I started by generating a TypeTray:

https://www.festi.info/boxes.py/TypeTray?language=en

When I generated the SVG, I see additional "chads" next to the finger joints in the "inner x" cuts, each time it intersects with the "inner y" trays.

I haven't cut this yet, but since there are no corresponding holes in the bottom plate, I don't see how the "inner x" walls would fit.

**To Reproduce**
TypeTray [default settings](https://www.festi.info/boxes.py/TypeTray?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=0.5&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&Hinge_grip_percentage=0&Hinge_outset=0&Hinge_pinwidth=0.5&Hinge_style=outset&Hinge_axle=2.0&Hinge_grip_length=0&Hinge_hingestrength=1&CabinetHinge_bore=3.2&CabinetHinge_eyes_per_hinge=5&CabinetHinge_hinges=2&CabinetHinge_style=inside&CabinetHinge_eye=1.5&CabinetHinge_play=0.05&CabinetHinge_spacing=2.0&Lid_angle=90.0&Lid_hole_width=0&Lid_second_pin=0&Lid_second_pin=1&Lid_spring=both&Lid_style=rectangular&Lid_surroundingspaces=2.0&Lid_edge_width=1.0&Lid_extra_length=0.0&Lid_finger=3.0&Lid_play=0.05&Lid_space=2.0&Lid_width=1.0&Click_angle=5.0&Click_bottom_radius=0.1&Click_depth=3.0&RoundedTriangleEdge_height=50.0&RoundedTriangleEdge_r_hole=2.0&RoundedTriangleEdge_radius=30.0&RoundedTriangleEdge_outset=1.0&Mounting_d_head=6.5&Mounting_d_shaft=3.0&Mounting_margin=0.125&Mounting_num=2&Mounting_side=back&Mounting_style=straight+edge%2C+within&HandleEdge_height=20.0&HandleEdge_hole_height=75.0&HandleEdge_hole_width=40%3A40&HandleEdge_on_sides=0&HandleEdge_on_sides=1&HandleEdge_radius=10.0&HandleEdge_outset=1.0&sx=50*3&sy=50*3&h=100.0&hi=0.0&outside=0&outside=1&bottom_edge=h&top_edge=e&back_height=0.0&radius=0.0&gripheight=30&gripwidth=70&handle=0&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&render=1) show this, but here's the [box I was creating](https://www.festi.info/boxes.py/TypeTray?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=0.5&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Stackable_angle=60&Stackable_height=2.0&Stackable_holedistance=1.0&Stackable_width=4.0&Hinge_grip_percentage=0&Hinge_outset=0&Hinge_pinwidth=0.5&Hinge_style=outset&Hinge_axle=2.0&Hinge_grip_length=0&Hinge_hingestrength=1&CabinetHinge_bore=3.2&CabinetHinge_eyes_per_hinge=5&CabinetHinge_hinges=2&CabinetHinge_style=inside&CabinetHinge_eye=1.5&CabinetHinge_play=0.05&CabinetHinge_spacing=2.0&Lid_angle=90.0&Lid_hole_width=0&Lid_second_pin=0&Lid_second_pin=1&Lid_spring=both&Lid_style=rectangular&Lid_surroundingspaces=2.0&Lid_edge_width=1.0&Lid_extra_length=0.0&Lid_finger=3.0&Lid_play=0.05&Lid_space=2.0&Lid_width=1.0&Click_angle=5.0&Click_bottom_radius=0.1&Click_depth=3.0&RoundedTriangleEdge_height=50.0&RoundedTriangleEdge_r_hole=2.0&RoundedTriangleEdge_radius=30.0&RoundedTriangleEdge_outset=1.0&Mounting_d_head=6.5&Mounting_d_shaft=3.0&Mounting_margin=0.125&Mounting_num=2&Mounting_side=back&Mounting_style=straight+edge%2C+within&HandleEdge_height=20.0&HandleEdge_hole_height=75.0&HandleEdge_hole_width=40%3A40&HandleEdge_on_sides=0&HandleEdge_on_sides=1&HandleEdge_radius=10.0&HandleEdge_outset=1.0&sx=27*6&sy=27*3&h=30&hi=25&outside=0&bottom_edge=h&top_edge=e&back_height=0.0&radius=0.0&gripheight=0&gripwidth=0&handle=0&thickness=3.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&render=1)

**Expected behavior**
Even though I haven't printed this yet, it doesn't look like the "inner x" cuts will fit into the bottom tray, since they have those additional chads.

**Screenshots or Drawings**
The red highlighted boxes below show the extra chads.  There are no matching holes in the bottom plate.

[image]

**Additional context**
Add any other context about the problem here.

closed 2022-11-12 by florianfesti

florianfesti, 2022-11-12:
Thanks for reporting! Fixed on the web instance now.

nicjansma, 2022-11-12:
Wow!  That was amazingly quick.  Thanks @florianfesti!

commit 6ab0cef4 (2022-11-12): Use straight edge for slots SlottedEdge

Used the connecting edge due to an aliasing issue. This broke as the
FingerJointEdge now places fingers even for smaller lengths.

Thanks to Nic Jansma (https://github.com/nicjansma) for reporting!

Resolves: #452