florianfesti/boxes #454: markdown installed by get error "No module named markdown" when running boxesserver

decon89, 2022-11-23:
When I try to run the boxesserver.py script I get the following error:

>  File "C:\Users\xxxx\Downloads\boxes-master\scripts\boxesserver", line 27, in <module>
>     import markdown
> ModuleNotFoundError: No module named 'markdown'

I expected this to work as I have installed markdown:

> pip install Markdown
> Requirement already satisfied: Markdown in c:\users\xxxx\appdata\local\programs\python\python311\lib\site-packages (3.4.1)

Is this a fault on my end or with the script? And how might I fix this?

florianfesti, 2022-11-28:
This may be an issue with multiple Pyhton installations. One having markdown installed and one running boxes.py. Try executing it with `python boxes` to use the python version in your path.

florianfesti, 2022-12-31:
I hope this fixed your issue! Closing.

closed 2022-12-31 by florianfesti