florianfesti/boxes #455: Unclosed fan hole swirls

ednisley, 2022-11-23:
The FanHole generator (`https://festi.info/boxes.py/FanHole`) doesn't quite close the tips of the swirl holes:

[image]

The missing segment seems to have become a tiny arc, which I found while using LightBurn to set up the SVG file for laser cutting. The red line in this image goes between the dangling end of the tiny arc and the tip of the swirl:

[image]

The missing segment should replace the tiny arc and join the swirl at the tip.

My Python-Fu is weak, but I think the problem lies in this section of `fanhole.py`:

```        for i in range(n):
            with self.saved_context():
                self.corner(-ao, r)
                self.corner(-a2)
                self.arc(d2, -90)
                self.corner(-180+a3)
                self.arc(d1, 85)
                self.corner(-90-a1)

            self.moveArc(-360./n, r)
```
Perhaps the `moveArc` has an incorrect endpoint or should be just another `arc`?

Also, the `CW Swirl` and `CCW Swirl` values in `boxes.py.FanHole.inx` require an additional quote layer to produce the proper quoted command-line string with the embedded blank. I added single quotes to move along, but perhaps using an underscore rather than a blank would be better:

```
        <param name="style" type="optiongroup" appearance="combo" gui-text="style" gui-description="Style of the fan hole">
            <option value="'CW Swirl'">CW Swirl</option>
            <option value="'CCW Swirl'">CCW Swirl</option>
            <option value="Hole">Hole</option>
```

The URL producing the swirl:

https://festi.info/boxes.py/FanHole?diameter=75&mounting_holes=3&mounting_holes_inset=1.75&arms=7&inner_disc=0.43&style=CW+Swirl&thickness=3.2&format=svg&tabs=0.0&debug=0&labels=0&reference=0&inner_corners=corner&burn=0.04&render=1

After a bit of manual repair, though, it turned out fine:

[image]

Thanks!

florianfesti, 2022-12-02:
Yeah, there is something wrong with the calculation of these arcs. I need to re-do them in a moment of great clarity.

ednisley, 2022-12-02:
> a moment of great clarity

A strong need for powerful caffeination! :grinning: 

Thanks …