florianfesti/boxes #456: Another console box, with somewhat irregular polygons - How to automatically connect/close a polygon?

lauri-aarnio, 2022-11-27:
I'm trying to create a console box, using the the "console.py" generator as the starting point. But I need a bit more complex shape for the sides than what it can generate now: There will be eight parts in total (six panels + the two sides), and the polygons which define the sides will have six different angles at different corners. Only one of those will be 90 degrees; the rest are more or less random.

Just now the polygonWall routine seems to require that the "borders" parameter defines all edges of the polygon. I.e. the last one should end at the same location were the shape started... and calculating that manually is a bit complex task, which needs to be repeated if/when the plan is changed.

I have been trying to read the source, to find out if there is already a routine which could calculate last edge for an incomplete polygon (but haven't found any solution). In other words, I'd like to define only five edges in "borders", and then call something to connect the first and last one to get a six-corner shape. Does something like that already exist?

Ranthalion, 2022-11-29:
I've searched through the project and don't see any functionality that would solve your issue.  

I think there are a couple of ways to accomplish this.  The first is to add an additional parameter to polygonWall to specify that the shape should be auto closed.  If this parameter is true, then a final segment will be calculated based on the location of the last point.

The other option would be to define the polygon based on a series of points. That would probably need to be a completely new method since it would be significantly different from the existing functionality.

Do you have a preferred approach?

lauri-aarnio, 2022-11-29:
After trying to calculate the polygon manually, I wrote a small program which generates the "borders" array. I also noticed that the best way is to have both possibilities: To be able to specify coordinates of some corners, and some need to be specified as angle+length.

The reason is that my shape requires that one corner is exactly 123.5 degrees. Additionally, height of the final box is important, and that would be difficult to achieve if everything would be based on just angles and lengths. Instead, it is really simple to specify with coordinates.

So, my borders-generator starts with empty "borders" and the code looks like this:

add2borders_angle_len(0, 220);       # starts from (0,0), goes to (0,220)
add2borders_angle_len(180-123.5, 132);  # to get 123.5 degree angle, turn 180-123.5 = 56.5 degrees
add2borders_goto_x_y(215,305);    # inside height = 305mm
add2borders_goto_x_y(215-95,305);
add2borders_goto_x_y(0,90);
add2borders_goto_x_y(0,0);  # returns to the starting point.
flush_borders();

As you can see, I didn't do that with Python - I used Perl instead, just because I'm faster with that (last time when I did something with Python was in 2008, I think...) and then copied the results to the code derived from console.py. This was the result:

borders = [ 220.000000, 56.500000,
          132.000000, 55.272226,
          209.900144, 68.227774,
          95.000000, 60.832387,
          246.221445, 29.167613,
          90.000000, 90]

To me it seems that "flush_borders()" needs to add an angle (this is the last number in it) which points to the second point in the shape, but I'm not sure about that value. I experimented with manual modifications to it, to generate something which looks reasonable.

So, it looks like I'm able to generate what I need, but this is still far from a generic solution, unfortunately.

Ranthalion, 2022-11-30:
I'm glad to hear that you were able to solve your issue.  It's probably been about that long since I've touched Perl!
If you'd be willing to share your program, I'd like to incorporate these methods into a border builder to help in the future. 
I also agree that the final angle doesn't seem to matter since it is setting the direction for the next line.

lauri-aarnio, 2022-12-01:
Sounds great! I'll need to clean it up and do some additional testing, and after that I'll share it here. Probably after next weekend, I think.

florianfesti, 2023-03-04:
OK, adding `None` at the end of `borders` now closes the polygon. Code is still not that well tested. Please file tickets or re-open this one if there are issues. Adding a link to the broken code is advised.

commit 9a541098 (2023-03-04): .polygonWall(s): Allow None at the end to close the polygon

Resolves: #456
closed 2023-03-04 by florianfesti