florianfesti/boxes #458: polygonWall not starting from last point

Ranthalion, 2022-11-28:
**Describe the bug**
polygonWall seems to start a line at the wrong location when creating a polygon with 5 sides.

**To Reproduce**
Pull PR #457 and create a display shelf with slope_top selected.   
[image]

In my case, the calculated borders array evaluates as follows.
```
borders = [215.0, 90, 35.355, 45.0, 254.056, -135, -38.348, -90, 215.0, 90]
self.polygonWall(borders, edge='e', move="up")

```
**Expected behavior**
Each line should begin where the previous line ends.

**Screenshots or Drawings**
This screenshot shows the lines not starting where the previous ends.  This screenshot was created without the callback to add finger holes, but the issue is consistent regardless of callback.
[image]

Also, there seems to be an unexpected nub here.
[image]

**Additional context**
This bug was discovered while attempting to add additional parameters to display shelf for including a back panel and sloping the side walls at the top.

Ranthalion, 2022-11-28:
Upon closer inspection, I found that the "nub" is actually a little arc.
[image]

florianfesti, 2022-11-28:
See https://florianfesti.github.io/boxes/html/api_burn.html

Ranthalion, 2022-11-28:
Burn Correction does seem to be related.  Changing it to 0 gave better results.  
I've also change from negative angles to only use positive angles.  That has gotten me closer to an answer.  My polygon isn't closed, but that may be due to rounding errors.  I'll continue with my implementation and close this issue (and draft PR) if these turn out to be the root causes of my issues.

Ranthalion, 2022-11-28:
Confirmed that root cause was a negative segment length.  Closing.

closed 2022-11-28 by Ranthalion