florianfesti/boxes #464: DisplayShelf Enhancement: Slope the top and include cord holes

Ranthalion, 2022-12-06:
I would like DisplayShelf to be enhanced with a few new options.
1. Include holes to run a power cord
2. Cut the top of the side walls based on the shelf angle.
3. Create a back wall

[image]

Ranthalion, 2022-12-06:
I've started implementation for this feature by adding a few new parameters.
[image]

florianfesti, 2022-12-06:
Details like holes for cords should not be part of the generator. Just add them "by hand" in either a vector graphics program like Inkscape or the software controlling the laser cutter.

The other two features make a lot more sense.

florianfesti, 2022-12-06:
As I see these neat French cleats in the background: Are you interested in testing some of the "Wallmounted" generators with the new French cleat feature?

Ranthalion, 2022-12-06:
> As I see these neat French cleats in the background: Are you interested in testing some of the "Wallmounted" generators with the new French cleat feature?

I'm willing to help with some of these, just let me know or tag me in an issue.

On the other hand, some of my other personal prioritized ideas are:
1. Automated regression/snapshot test on generators with default parameters
2. Investigate python to WASM to reduce server resources and generate previews in real time
3. Validation/Sanity check messages (For example, overall height of 100mm, but a shelf height of 200mm)
4. New generator - bent acrylic easel stand. Similar to 
[image]

PR #466 (2023-01-01): DisplayShelf: Implement back and slope
Implement Sloped Sides and Back Wall for DisplayShelf.

Fixes #464
closed 2023-01-01 by florianfesti