florianfesti/boxes #467: Stabilizers and/or Bottom Insets

xloem, 2022-12-19:
This project is such a boon to exist, and such a relief to find. I needed this years ago.

I noticed the paint can shelves include stabilizers to fit inside the stackable feet. It would be nice if these stabilizers were available in other stackable boxes.

With stabilizers taller than the bottom edge, one could possibly make boxes that stack with commercial milk crates here in the USA. Milk crates stack by having an inset in the bottom that is smaller than the walls.

This could also be accomplished with something like the underside of a lid alongside the bottom, maybe a couple layers, or with a hollowed inside to save weight.

florianfesti, 2023-01-02:
Have a look at the `bottom_stabilizers` parameter in StackableEdgeSettings and `bottom_lip` in `FingerJointSettings` added with b5d10334d0955c8707b70424c7192862d709b1de and 4d41b2ddb1a70106f090136590171674a1b81bd6

Please test and close this issue on success. If you have pictures of the details they may become handy in the documentation.

florianfesti, 2023-01-14:
Guess this is simple enough to just close.

closed 2023-01-14 by florianfesti