florianfesti/boxes #469: TypeTray renders rectangular slide on lids incorrectly.

markwal, 2022-12-21:
**Describe the bug**
TypeTray renders the slide on lid "L" incorrectly.  I believe this is a result of the fix for #422 (93a4cad) which appears to be incorrect.  I can give a PR that switches them back while maintaining the new naming, but I'd like to know if the fix (93a4cad) had a test case that it fixed or if it was based on the mentioned reverse engineering only?

**To Reproduce**
boxes TypeTray ---format=svg --outside=0 --sx="250:27" --sy="93:93:40" --gripwidth=0 --top_edge="L"

In particular, it is the top_edge="L" with a rectangular tray.

**Result**
The lid left, right, front, back doesn't match with the wall, left, right, front, back.

**Expected behavior**
The lid to match the walls.

florianfesti, 2023-01-16:
Yes, easily reproduced. Turns out this is all mess up completely. Looks like someone (might have been me) has redone the top edges under the assumption that UniversalBox uses them correctly. But turns out UniversalBox used y, x, h instead of x, y, h. So it is not messed up for everything else. This needs a look at the whole thing to disentangle this mess.

w1ml33na3rts, 2023-04-01:
Is this going to be fixed any time soon or should I try to work around the issue as mentioned in issue #502 ?

florianfesti, 2023-04-01:
OK, this should be clean up now. Please report back if anything is still ajar.

closed 2023-04-01 by florianfesti

w1ml33na3rts, 2023-04-02:
> OK, this should be clean up now. Please report back if anything is still ajar.

Wow, great service Florian, thanks! It certainly looks correct. I'll try to cut one Thursday and come back here if things aren't right yet.