florianfesti/boxes #485: cajón con repisa

CARRION85, 2023-01-04:
[image]
By suggesting a new box generator or other feature you agree to test cut it until it works. You understand that this may take a few attempts. As soon as it does work you will provide a sample image that can be used in the user interface or the documentation.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you think would be the best solution.

** Pictures of something similar **
If possible provide a picture of a similar item or feature. If copy right allows it uploading it to this ticket is preferred as it is less likely to disappear.

CARRION85, 2023-01-04:
Quiero hacer este tipo de cajones . Se podría hacer un generador ?

florianfesti, 2023-01-05:
Have a look at the TypeTray generator. If you select the `handle` check box the bottom is extended to the front just is in your picture. You can just delete the grip holes and adjust  this part in Inkscape.

closed 2023-01-05 by florianfesti

CARRION85, 2023-01-06:
ok. lo intentare. gracias