florianfesti/boxes #486: Presets for specific products

patvdleer, 2023-01-05:
How about adding presets of settings for specific boxes? Take [CanStorage](https://www.festi.info/boxes.py/CanStorage) for example, a preset for AA, AAA, 18350, 18650 and such would be really helpful.

florianfesti, 2023-01-31:
Yeah, I have been wondering about that, too. Question is how to do that easily and without cluttering the UI too much.

I could imagine an optional drop box and "Load defaults" button on the right aligned on the "Generate" button row. As we can pass parameter in the URL this should actually not be too complicated to implement. May be a bit infrastructure is needed to make this convenient API wise.

patvdleer, 2023-02-01:
I was thinking about creating a bunch of classes define default shapes like cylinders, extend on those to create a category for batteries and create a list of common types, AA/AAA/18650. For hooking these up to the models I thought about creating a mapping, so for example the diameter of the battery links to some field of the model thus knowing what field to set.

So I was thinking about playing around with it but looking at the boxesserver I got a headache and went a little overkill. I rewrote the web server in Flask (not complete overkill, I could have gone with Django). This with the goal of creating an interface that is more manageable than the boxesserver. I'm planning on seeing if I can define the defaults in json and expanding the forms to generate the models. Was thinking about using a modal where the model can be selected and optionally a margin. 

Tell me if I'm going overboard (or when I did...)

gcollic, 2023-02-01:
It would definitely be usefull for a lot of things ❤️ 

Some other ideas:

- Phone models for [PhoneHolder](https://www.festi.info/boxes.py/PhoneHolder)
- Keyboard [layouts](https://github.com/help-14/mechanical-keyboard) for [Atreus21](https://www.festi.info/boxes.py/Atreus21) and [Keypad](https://www.festi.info/boxes.py/Keypad)
- Card sizes for [DividerTray](https://www.festi.info/boxes.py/DividerTray), [CardBox](https://www.festi.info/boxes.py/CardBox), [CardHolder](https://www.festi.info/boxes.py/CardHolder)
    - Mini American: 41×63mm
    - Mini European: 44×68mm
    - Standard American: 57×89mm
    - Standard European: 59×92mm
    - Standard Card Game: 63.5×88mm
    - Tarot: 70×120mm
- Notes sizes for [PhoneHolder](https://www.festi.info/boxes.py/NotesHolder)
    - Classic: 76×76mm
    - Wide: 76×127mm
- Ikea and other furnitures models for [StorageShelf](https://www.festi.info/boxes.py/StorageShelf)
- Standard computer fan sizes for [FanHole](https://www.festi.info/boxes.py/FanHole)
    - 40mm
    - 80mm
    - 92mm
    - 120mm
    - 140mm
    - 180mm
    - 200mm

patvdleer, 2023-02-01:
I love it! now to figure out the best way to get there. The backend part shouldn't be that difficult, the frontend might be a bitch...

https://github.com/patvdleer/boxes/tree/flask , code is in the folder called web, might be horribly overkill but it seems to be a lot easier to update/change than the current script (or at least for me it is)

florianfesti, 2023-02-01:
The back end part is the thing to get right. May be just adding a presets property to the Boxes class is enough. It could be a dict "Description" -> settingsdict. With the settingsdict just being "settingsname" -> value.

The front end would need to turn that into a dropbox that invokes a link to the generator with the preset values in the URL. If this is too complicated it can just give the name of the presets and the Boxes class has to load them in (probably) `parseArgs`. The later even had the benefit of keeping all changes but the presets.
This sound nearly too easy...

ccrome, 2023-02-01:
Maybe add to the list:

* Screw sizes, metric and 'murkin
* bearings: IDs and ODs

There are so many possibilities.  I think getting the form of the database right is key, then people could add whatever kind of items they want.

patvdleer, 2023-02-01:
@ccrome the reason I was thinking about a (more elaborate) class based system is specifically screws and such. The step after default simple products would be to have dynamic or combined/compound product like screws with different heads or more complex objects were there are a lot of sizes of. It would allow us to extend on other types and use property functions rather than a "hard-coded" dict of measurements.

So rather than just a dropdown in the interface you would also be able to select a type of object and enter your own base measurement. In stead of writing out all types of screw heads and generate 1M up to 20M a user could select the head type from the dropdown and enter the metric value letting the model calculate the correct dimensions. 

I hope Im getting making any sense

patvdleer, 2023-02-03:
a small brain fart:

basic shapes:
```python
import math
from dataclasses import dataclass

@dataclass
class Circle:
    diameter: float

    @property
    def radius(self) -> float:
        return self.diameter / 2

    @radius.setter
    def set_radius(self, radius) -> None:
        self.diameter = radius * 2

    @property
    def circumference(self) -> float:
        return math.pi * self.diameter

@dataclass
class Cylinder(Circle):
    height: float

    @property
    def volume(self) -> float:
        return math.pi * (self.radius**2) * self.height
```

battery example:
```python
class AA(Product, Cylinder, Battery):
    name = "AA"
    height = 50.5
    diameter = 14.5

class AAA(Product, Cylinder, Battery):
    name = "AAA"
    height = 44.5
    diameter = 10.5
```

OR

```python
class DryCell(Product, Cylinder, Battery):
    pass

batteries = [
    DryCell(name="AA", height=50.5, diameter=14.5),
    DryCell(name="AAA", height=44.5, diameter=10.5)
]
```

patvdleer, 2023-02-03:
A quick attempt: https://github.com/patvdleer/boxes/tree/product_measurements

HaSHsss, 2023-02-03:
i see it like this 
We can create custom JSON (with required preset settings) files and store them next to the generator.
In the presence of such files, display buttons that, when pressed, will redirect to a page with already filled data.
If we use a dropdown list, you will need to use Javascript etc. No problem for me, but i think, lot of users here don like it.

Something like file `Abox-cubebox.json`
`box  :  ABox{
	FingerJoint_angle : 90.0
	FingerJoint_style : rectangular
	FingerJoint_surroundingspaces : 2.0
	FingerJoint_bottom_lip : 0.0
	FingerJoint_edge_width : 1.0
	FingerJoint_extra_length : 0.0
	FingerJoint_finger : 2.0
	FingerJoint_play : 0.0
	FingerJoint_space : 2.0
	FingerJoint_width : 1.0
	x : 100.0
	y : 100.0
	h : 100.0
	outside : 0
	outside : 1
	bottom_edge : h
	thickness : 3.0
	format : svg
	tabs : 0.0
	debug : 0
	labels : 0
	labels : 1
	reference : 0
	inner_corners : corner
	burn : 0.1
	render : 0
}`

patvdleer, 2023-02-03:
@HaSHsss how do most users use Boxes? Via webinterface no? As much as I love the terminal it seems like a pain to do this on a bare terminal (as in, without TUI)

HaSHsss, 2023-02-03:
> @HaSHsss how do most users use Boxes? Via webinterface no? As much as I love the terminal it seems like a pain to do this on a bare terminal (as in, without TUI)

I meant developers under the word `user`.
It doesn’t matter to end users how it works. I mean, we have 1 developer here, this is @florianfesti . And we are just `advanced users` of hims product. 
IMHO

HaSHsss, 2023-02-03:
> how do most users use Boxes? Via webinterface no? As much as I love the terminal it seems like a pain to do this on a bare terminal (as in, without TUI)

bare terminal (as in, without TUI) - what are you talking about. I didn't suggest it

gcollic, 2023-02-04:
I also lean towards simple preset of settings which are just default values, but I'm not sold on the JSON way. Every boxes have different settings, and it would be tedious to add another item of a standard kind to multiple boxes compatible with it. If the exact same presets are directly in Python , you could just declare preset the same way as in JSON for simple cases, but it also makes it possible to concatenate mapping from multiple collection. For example, the NotesHolder could iterate from the PostItNotesStandardsSize [{name, width,height}] to generate some presets, concatenated with presets generated iterating from PlayingCardDeckStandardSize [{name, width, height, depth}], and as soon as a new size is added in a collection source it will impact the presets of every boxes using it.

patvdleer, 2023-02-05:
https://github.com/patvdleer/boxes/tree/flask_presets

[image]

PhoenyxAshe, 2023-05-02:
I don't need a lot of fancy presets... but could we add a diameter input on the Coffee Capsule Holder?  I was able to re-size it for K-cups finally, but it was quite a pain.

gcollic, 2023-05-02:
> I don't need a lot of fancy presets... but could we add a diameter input on the Coffee Capsule Holder? I was able to re-size it for K-cups finally, but it was quite a pain.

I think @florianfesti suggested the same thing when I contributed this "box". @PhoenyxAshe Do you have the hard-coded settings values you changed for K-cups ? I can add the input, with the settings value in the "box" description until preset exists.

PhoenyxAshe, 2023-05-02:
I don't. The first one I made was a number of... eventful... months ago.  I think I set the values in the generator at 3 rows... 4 or 5 capsules by row, and when I cut the pieces, they were too small.  So in Inkscape I duplicated the top panel, then chopped it down to just part of one of the rows - basically the opening plus enough for a bit more than one capsule past so I could make sure that the capsule would both fit into the opening and into the rest of the holder.  Then I started scaling up, cutting just that test piece, and scaling again until I got the right size.  Looking at my files, it looks like it took four tries. (Original file was 01, followed by 01a-01d.)  Again, memory is spotty on the details, but I'm pretty sure I measured the cups around that time as well, both the upper diameter of the cup itself and the lip, and measured in Inkscape as well, so it was mostly a matter of fine tuning.  

For the generator itself, the only presets are "columns" and "capsules by column".  Therefore, the only way to change it for different capsules is after the file has been generated.    Adding the additional inputs of "opening diameter" and "capsule diameter" would make the generator a _lot_ more flexible, enabling it to do more than just coffee capsules.  I'm not sure why my brain went in that direction, but with the right thickness of wood or extra layers, wine glasses could even be done, like in the image I just grabbed:

[image]

Ranthalion, 2023-07-01:
@patvdleer I'd love to see flask rolled in to boxes!  I don't see an issue or PR for it.  I'd be happy to help if there are any gaps left.

patvdleer, 2023-07-01:
@Ranthalion I didn't create one, I've started work on https://github.com/patvdleer/boxes/tree/flask_presets but due to personal life I haven't found the time to continue working on it.

Ranthalion, 2023-07-02:
@HaSHsss  I like the idea of presets being represented by json files.  I think this could open up a larger presets ecosystem if we support loading presets from a url.  

Users could publish their own presets to any github account (or anywhere web accessible, I guess).  Boxes could have an official presets repo, but users could copy/paste the url for repos with a collection of presets, or the url for a single preset.

bbunch1970, 2024-12-08:
> Coffee

I found at least for the cups I had it was way small and had to resize, really need to more variables, cup diameter and rim diameter, rim diam would set the center section and the enlarged holes on end of top section's slots and cup diam would set the top section's slot size.,  I made further mods on the one I ran, I ran 2 top sections so that the cups can put in either way (bottom one for ex can be label out to see what is in the column and the other where you can grab them from the front)and will be mounting to shelf brackets so it can be set on a counter and weighted or clamped down.  I wanted to try this project and donating the trial run to the farmer's market for their coffee stand.

bbunch1970, 2025-11-02:
> I don't need a lot of fancy presets... but could we add a diameter input on the Coffee Capsule Holder? I was able to re-size it for K-cups finally, but it was quite a pain.

I had suggested same sort of ieda, rim and cup diameter as variables so you could really set for about any cup with a lip.