florianfesti/boxes #489: Gridfinity Generators

ccrome, 2023-01-06:
**Describe the solution you'd like**
The [gridfinity system](https://www.youtube.com/watch?app=desktop&v=ra_9zU-mnl8) is great, and has a large [open-source following](https://github.com/jeffbarr/gridfinity-catalog).  I'd like to add a gridfinity base plate generator, as well as gridfinity modules as well.  

I have created a[ pull request for the base plate generator here.](https://github.com/florianfesti/boxes/pull/487).  Sorry if I'm doing things out of order with regard to pull requests and issues.  

[image]

Additionally, I'd like to add shelves to match the trays in the above pull request.  We can use any of the shelving systems that already exist, but that requires knowing the exact dimensions needed.  My thought is to have a simple shelf system that will be based on the number of grid units, rather than physical dimensions.  Basically, I want to say, "generate a gridfinity shelf system to house 4x8 grid trays" .  I haven't thought through the full shelving system yet. Starting with just the trays for the moment.

** Pictures of something similar **
The attached pictures are generated by the code in the above pull request.

florianfesti, 2023-01-08:
If there are more than just two or three generators we can put them into their own UI group. Otoh Gridfinity with all the 45° surfaces does really lend itself to laser cutting. So we might be stuck at just trays and shelves.

ccrome, 2023-01-08:
I was wondering about that.  The 45 degree surfaces aren't always necessary, so I think there's some opportunity here for lasering the bins themselves.  3d printing is great, but it's just so slow.

I created a new type of shelf system for my trays, with adjustable shelves.  It's not quite done yet, but you can have a [look at it here](https://github.com/ccrome/boxes/tree/add-adjustable-shelf) on the 'adjustable-shelf' branch.

It's a lot like the DividerTray, but is really meant to work more as a shelving system with movable shelves, and to be sturdy.  I haven't cut a large one yet, but here's my cute little baby one
[image]

It does use a good bit more material for the shelf slides, but this material I'm using is so cheap, it really doesn't matter.  It's like $12 for a 4'x8' sheet (about 1.5 square meters).  

The code is fairly dumb right now -- since I know how to make holes, that's what I'm doing, rather than drawing out the proper shapes.  But it works for the first tests.

florianfesti, 2023-01-11:
I wonder if regular (Type)-Tray can be adjusted using a `h` bottom edge and the new `bottom_lip` feature added with #467 
Those could grow slots to fit into the grid.

florianfesti, 2023-01-11:
One could also adjust the TrayLayout. That would make a lot more fun if one didn't have to type in dimmensions.

florianfesti, 2023-01-16:
Other idea: Have a box with shortened fingers is the finger joints. This should create a fold all way round.

ccrome, 2023-01-16:
I was playing with the TypeTray, and it's pretty awesome.  It seems like that can be the basis for a whole lot of things.  I starting building a large shelf system (about 450mm high), and realized the 3mm sheet is just really flimsy.  Rookie mistake.  :-)

I'm not quite sure what you mean by a "fold" all the way around.

florianfesti, 2023-01-17:
Here the cross section without and with a fold

[image]

florianfesti, 2023-02-17:
Ok, any other idea for Gridfinity generators you want to work on? Or can we just close this ticket?

Hmm, the new SmallPartsTrays could be interesting...

ccrome, 2023-02-17:
> 

Yes, we can close the ticket.  But yes, there are lots of options for generators.  Not sure if we should just leave this open for future discussion or just close it.  I don't care either way.

Yeah, I was thinking about the SmallPartsTray too.

florianfesti, 2023-02-18:
I think we can do discussions in the PRs or - if necessary - in new tickets for specific ideas. Closing.

closed 2023-02-18 by florianfesti