florianfesti/boxes #497: Add more trays for game tokens

florianfesti, 2023-01-09:
We already have a few board game inserts but there is clearly more to be added.

First thing is the new [SmallPartsTray](https://festi.info/boxes.py/SmallPartsTray?language=en) (still untested and though still in unstable). @gcollic interested in having a look?

A similar thing with different presets could be used for hex tiles.

gcollic, 2023-01-09:
I'm loving it!

> @gcollic interested in having a look?

Do you have something particular in mind? I could do a test cut and a picture of the smallpartstray for example. I have less time right now to contribute full new components sadly.

florianfesti, 2023-01-09:
I am actually even more interested in your opinion and suggestions. Although you holding one in your hand can probably not hurt with that.

I also added the obviously missing `outside` parameter.

Overall plan is to get a few stand alone trays for board games and then finally get #234 done to make it easier to create inserts for whole games. So people with little time can still do them ;)