florianfesti/boxes #507: web server versioning

ccrome, 2023-01-17:
Is this a dumb idea?
[image]

You can check it out live here: http://boxes.crome.org.

And the commit is here: https://github.com/ccrome/boxes/commit/fb8ccf2ef31763400d2beb1d3546384566e5d14a

I'm getting the git info via subprocess.run() as opposed to strictly pythonic git just so it doesn't require more dependencies.  Not sure if that's the right way to go.

florianfesti, 2023-01-31:
This is not a dumb idea. I not quite sure if it is the right solution. For one this should go into the meta data of the drawings. I doubt anyone will write down the version SHA1 when generating a box. Then there is the question what happens when Boxes.py is run from a proper installation (no idea if anyone really does that). For that setup.py needs to drop the SHA1 into some file and ship that. Not sure if that's worth the trouble.

ccrome, 2023-01-31:
I was thinking it would go as a URL parameter, and so get stored in whatever QR code was generated.  My eventual thought is that you can get back and pick up from where you left off, even when the server version has changed.

florianfesti, 2023-02-26:
Ok, here is how I think this should work:

There's a module in boxes/ that gets the git SHA1, branch, commit date etc. It falls back to loading another Python module that is generated using the same code during build in `setup.py`. That way packaged boxes versions also have the git info even if they are not in git.
The Boxes class uses this module to add the git info to its `.metadata` dict and the different `Surface` classes in `boxes.drawing` add those to the generated files.

That's probably not too complicated to do - especially when following the example of the short url and cli versions that got just added with  0421aeb76a2659d0657d6d4f86440bb7f5ae25a5 and afbad13f51c9c4c4cb636409aa4ee5cbee631dc9
Still that's basically a complete re-write of this feature. While I am happy to do the final tune up that's a bit too much to do on my own while the PR queue is still not empty and my own repo is full of unfinished branches...

florianfesti, 2024-03-23:
I don't see this happening any time soon. Feel free to re-open if you are working on this. Closing for now.

closed 2024-03-23 by florianfesti