florianfesti/boxes #516: No boxes configuration windows on Fedora 36

jgforbes, 2023-01-23:
Fresh install of Fedora 36 and Inkscape from Fedora repository.
Installed boxes from git using install script.
The boxes extension shows in the extensions pull-down; however,
clicking on one of them does not bring up a configuration window.
Inkscape complains about image files at startup:
Image file ('/usr/share/inkscape/extensions/./WineRack-thumb.jpg') not found for image widget in extension 'info.festi.boxes.py.WineRack'
but no errors when one of the box types is clicked.

Any ideas on what to check?

florianfesti, 2023-01-24:
Can you please try out daa3a50600fad16235f722f479f65c8413253c56 ? This won't fix the missing pictures but should make the generators work.

tiberiuichim, 2023-02-05:
@florianfesti I'm trying to get the inkscape extension to work.

I've cloned, run `sudo pip install -e .`, it gets installed in `/usr/share/inkscape/extensions`

Now when I enable preview on one of the boxes, I get:

```
Traceback (most recent call last):
  File "/home/tibi/work/boxes/boxes/drawing.py", line 572, in finish
    tree.write(open(self._fname, "wb"), encoding="utf-8", xml_declaration=True, method="xml")
PermissionError: [Errno 13] Permission denied: 'box.svg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/boxes", line 7, in <module>
    exec(compile(f.read(), __file__, 'exec'))
  File "/home/tibi/work/boxes/scripts/boxes", line 105, in <module>
    main()
  File "/home/tibi/work/boxes/scripts/boxes", line 45, in main
    run_generator(name, sys.argv[2:])
  File "/home/tibi/work/boxes/scripts/boxes", line 71, in run_generator
    box.close()
  File "/home/tibi/work/boxes/boxes/__init__.py", line 716, in close
    self.surface.finish(self.inner_corners)
  File "/home/tibi/work/boxes/boxes/drawing.py", line 574, in finish
    raise ValueError((self._fname, os.getcwd()))
ValueError: ('box.svg', '/usr/share/inkscape/extensions')
There was no box.svg output generated. Cannot continue. Command was:
boxes abox --x 100 --y 100 --h 100 --outside true --bottom_edge F --thickness 3 --tabs 0 --debug false --labels true --reference 100 --inner_corners loop --burn 0.1 --FingerJoint_angle 90 --FingerJoint_style rectangular --FingerJoint_surroundingspaces 2 --FingerJoint_bottom_lip 0 --FingerJoint_edge_width 1 --FingerJoint_extra_length 0 --FingerJoint_finger 2 --FingerJoint_play 0 --FingerJoint_space 2 --FingerJoint_width 1 --output box.svg box.svg

```

The `ValueError` is my own debugging, it prints `self._fname` and `os.getcwd()`. It seems that the extension wants to generate in the global extensions folder. I initially thought that it's due to trying it in a new Inkscape window (no saved file), but apparently it doesn't make a difference.

jgforbes, 2023-02-07:
I have the same issue with the latest version. Setting the extensions directory to world writeable allows it to work, but it should be writing the file  to /tmp instead.

florianfesti, 2023-02-08:
OK, I hope this fixes it. mkstemp to the rescue. No idea why that wasn't done from the beginning...

commit 8b961797 (2023-02-08): boxes_proxy (for Inkscape): Use temp file

Resolves permission issues when trying to place files into system extension
directory.

Resolves: #516
closed 2023-02-08 by florianfesti