florianfesti/boxes #532: Memory overflow

HaSHsss, 2023-02-05:
**Describe the bug**
If you enter a very large number in the parameters, then the script, trying to generate the layout, overflows the memory and the server CPU

**To Reproduce**
Run This URL in 5-10 tabs in parallel

https://www.festi.info/boxes.py/ABox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&x=99999999999999999999999999999999999999999999999999999999999999.0&y=99999999999999999999999999999999999999999999999999999999999999.0&h=99999999999999999999999999999999999999999999999999999999999999.0&outside=0&outside=1&bottom_edge=h&thickness=9999999999999999999999999999999999999999999999999999.0&format=svg&tabs=0.0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=99999999999999999999999999.000000000000000000001&render=1

**Expected behavior**
Need to limit the number of characters on the server side. Also need to take care to the characters after the dot

**Screenshots or Drawings**
[image]

[image]

https://www.festi.info/boxes.py/ is DOWN NOW.
I am sorry

florianfesti, 2023-02-05:
Deleted link as it randomly creates 100% CPU processes on the server.

HaSHsss, 2023-02-07:
**I will not give examples.**
I can enter a command and execute it using the input fields.

I have not been able to overwrite the executable file yet, but i can already edit the picture at this moment.

It is necessary to check in the incoming parameters that this is a number, Filter double or more spaces (I'm bad at Python, if I could write code in one line I could get around these restrictions), forbid special characters like (:)

HaSHsss, 2023-02-07:
I succeeded because I know the names of the variables

florianfesti, 2023-02-07:
Can you just send the reproducer to my commiter's email?

HaSHsss, 2023-02-07:
I can't send private messages here
 and I don't know your email

HaSHsss, 2023-02-07:
Just need to add  strip("/\_=+:)(")
or 
Something more powerful in regexp

HaSHsss, 2023-02-07:
```
input =  """
				<input name="%s" maxlength="7" type="number" value="%s">
```
in `boxesserver  `as minimal step

HaSHsss, 2023-02-07:
Oh 
input edge_types have text as valeu in 
https://festi.info/boxes.py/UnevenHeightBox

florianfesti, 2023-02-07:
Today has been a long day. Can you just send the URL(s) to florian at festi.info , please?

florianfesti, 2023-02-07:
455f5ae69a20c14d6da49b82baaa4c2859baed63 should at least protect against overly large numbers.

florianfesti, 2023-02-19:
> Just need to add strip("/_=+:)(") or Something more powerful in regexp

Well, first these should actually be treated properly by quoting/unquoting them. I could not quite see where exactly they slip through the cracks by looking at the code. Also these characters are actually used in the `sx`, `sy`, `sh` parameters and in the layout descriptions of `TrayLayout` and the new `GridfinityTrayLayout`. So filtering them is not an option.

> ```
> input =  """
> 				<input name="%s" maxlength="7" type="number" value="%s">
> ```

Well, `type=number` is for integers while most inputs are floats. And we make use of that at some very important places like `thickness` and `burn`. So going to ints is not an option.

florianfesti, 2023-03-11:
I am really not seeing how any of these special chars should be able to make it through argparse. Without a reproducer this is not going anywhere. Closing.

closed 2023-03-11 by florianfesti