florianfesti/boxes #534: Broken PDF file

xsurixxx, 2023-02-06:
**Describe the bug**
When you create your Integrated Hinge Box, and select the type of file you want download, if was PDF, when you download it, the file is broken, you just recieve some parts of the box

**To Reproduce**
Create a box and download on PDF format

**Expected behavior**
A complete file with entire box

**Screenshots or Drawings**
[image]

florianfesti, 2023-02-07:
Yes, easily reproduced. This is an issue with `pstoedit` whihc is used to convert the postscript Boxes.py actually is able to create to PDF (and several other formats). I could be this worked at some point... but I could not create a usable PDF with any option I tried (most didn't do anything). If anyone can provide a working `pstoedit` command line this would be an easy fix...

xsurixxx, 2023-02-08:
Funny, becouse i used on my work a lot of times, and worked...  One curious fact is when you download a project in PDF it come with exact measurements of the box you put on site, exemple: 100x100x100, in other files like PLT, i need put 40% more of the real mesure i need to obtain the correct file to cut on laser... Whatever, i apreciate your reply, i'll share this with my programmer friends to find some resolution for this issue ;D

Thanks and have a nice day!

xsurixxx, 2023-02-08:
Hii! Just a quickly update of this issue, now looking my old files i've downloaded on my work, all them are broken for some reason, it's so strange, maybe the issue is not on boxes.py, i don't know...

florianfesti, 2023-02-12:
I wonder if the behavior of the viewers have changed to now take the viewport into account instead of just showing everything that is drawn.

florianfesti, 2023-02-23:
ps2pdf -dEPSCrop <input.ps> <output.pdf> preserves the page size if the first line is changed to 

`%!PS-Adobe-2.0 EPSF-2.0`

Not yet sure if this is a good thing to do in general.

florianfesti, 2023-02-24:
Great! the AI format has the same issue - except there is no ps2ai utility...

florianfesti, 2023-02-24:
Ok, the PDF issue is solve with b1af53eff0e379a607a4f0e9b9d5476d8a26b5e3

But it silently changes the PS format to EPS. Not sure if this break anything else. Testing is appreciated before I put this up on the server.

florianfesti, 2023-02-28:
OK, there is `ps2ai.ps` in Ghostscript that is used by `pstoedit`

```
%         - To Do List -
%  find real %%BoundingBox: llx lly urx ury
...
```

```
(%!PS-Adobe-2.0 EPSF-1.2\n%%BoundingBox: 0 0 612 792\n) jp
(%%Title: Adobe Illustator 3 Editable Document\n) jp
(%%Creator: ps2ai.ps vers.) jp vers jpnum ( \(C\) 1993-94 Jason Olszewski\n) jp
(%%TemplateBox: 0 0 612 792\n) jp
```
Great!

florianfesti, 2023-03-12:
Basically closed as WONTFIX for the AI format. If anyone want to dig themselves through the AI reference manual it is technically probably not difficult to fork the PSSurface to an AISurface class. For me that's currently just not worth it.

commit cb449b87 (2023-03-12): Drop suppport for ai files

as the bounding box is not correct and other formats like ps and pdf are
very similar and work fine (now).

This should be revered in case ps2ai.ps properly processes the
BoundingBox. Given the fact that it is a PS script itself I don't quite
see how it wants to get access to a comment that's even later in the file. So I don't expect that to happen soon - or at all.

Alternative we could be producing AI files ourselves. It's just a gloryfied
PS file anyway. But that currently not planned.

Resolves: #534
closed 2023-03-12 by florianfesti