florianfesti/boxes #538: /bin/sh:1:boxes: not found

alexmlucas, 2023-02-13:
Still an issue with latest revision as described in bug #537

florianfesti, 2023-02-13:
Hmm, this looks like either `/usr/bin/env` or `python3` is not found. Can you check for both of those, please? I don't have a Debian installation at hand but I have a hard time believing they are not available. May be you don't have python3 installed?

alexmlucas, 2023-02-14:
So i've looked and /usr/bin/env exists. Also Python 3.9.1 is installed.

florianfesti, 2023-02-14:
Yeah, but is there a python3 binary in the path? If not, may be there is some python3 meta package containing a link. Or you need to adjust the first line of `boxes_proxy.py`

florianfesti, 2023-02-24:
Anyway, this doesn't really look like an issue with boxes.py. Closing for now. Feel free to re-open if there are more issues.

closed 2023-02-24 by florianfesti