florianfesti/boxes #541: TrayLayout broken?

caswal, 2023-02-21:
**Describe the bug**
Traylayout at https://www.festi.info/boxes.py/TrayLayout?language=en appears to be broken, and doesn't render the output box correctly.

**To Reproduce**
Go to, https://www.festi.info/boxes.py/TrayLayout?language=en, leave default x 2, y 2 arguments. Generate. Leave default settings, Generate.

Produces a collapsed 0 height? Set of sides, no labels. 

**Expected behavior**

Generates box svg drawing correctly, with labels. 

**Screenshots or Drawings**
[image]

[image]

**Additional context**
Using Chrome, Version 110.0.5481.104 if that is related.

Is a fantastic site.

florianfesti, 2023-02-21:
This is expected behavior. You need to add the sizes for the rows and columns in the `layout` field.

Note that this may change soon if PR #529 gets merged. I have not yet decided on how exactly the UI will look. This will be one of the next things I'll look into after #488 got merged.

caswal, 2023-02-21:
Ahh, that is quite non-obvious, to my tired just got a puppy brain.

Another option that could be nice, is the way some UI designers work. That you specify an overall size, and each row and column has a fill value. All fill values are summed and then normalized by the total.

[image]

So a 60/40 split on the columns.
A 0.25, 0.5, 0.25 split on the rows. Total = (0.5+1+0.5) = 2. So the rows are 0.5 / 2, 1 / 2,  0.5 / 2

ccrome, 2023-02-21:
You can play around a bit with https://www.festi.info/boxes.py/GridfinityTrayLayout?language=en.  It's based on the TrayLayout, and it auto updates the widths.  But I gives fixed external sizes only.

caswal, 2023-02-21:
Great thanks.

florianfesti, 2023-02-26:
Don't think we still need an open ticket for this. Closing.

closed 2023-02-26 by florianfesti