florianfesti/boxes #556: traceback on applying to generate a new box

bitbasher, 2023-04-30:
**Describe the bug**
i am in Mac OS X 11.7.6 Big Sur with Inkscape 1.2.2 installed
I used brew to install everything per the instructions on the installation page and all seemed well. I did get build warnings about setup.py and easyinstall being deprecated .. all of the dependencies reported that they were installed.
I went ahead to see if the extension would show up in Inkscape.
Yesterday evening i selected a simple divided box and generated it .. no problems.

**To Reproduce**
Today I wanted to try making a misc/hook drawing so opened a new document and then selected the generator .. changed some of the settings and tapped apply. I got a popup with this in it

> /bin/sh: boxes: command not found
Traceback (most recent call last):
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/boxes_proxy.py", line 76, in <module>
    boxesPyWrapper().run()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/base.py", line 231, in run
    self.save_raw(self.effect())
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex/extensions.py", line 234, in effect
    fragment = self.generate()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/boxes_proxy.py", line 65, in generate
    doc = etree.parse(stream, parser=etree.XMLParser(huge_tree=True))
  File "src/lxml/etree.pyx", line 3536, in lxml.etree.parse
  File "src/lxml/parser.pxi", line 1896, in lxml.etree._parseDocument
  File "src/lxml/parser.pxi", line 1916, in lxml.etree._parseFilelikeDocument
  File "src/lxml/parser.pxi", line 1811, in lxml.etree._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 1201, in lxml.etree._BaseParser._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 615, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 725, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 654, in lxml.etree._raiseParseError
  File "/var/folders/rs/ywn4bt8j3z7191g38d70fh740000gn/T/boxes.py-inkscapehwdhrhuy.svg", line 1
lxml.etree.XMLSyntaxError: Document is empty, line 1, column 1
sys:1: ResourceWarning: unclosed file <_io.TextIOWrapper name='/var/folders/rs/ywn4bt8j3z7191g38d70fh740000gn/T/boxes.py-inkscapehwdhrhuy.svg' mode='r' encoding='UTF-8'>

**Expected behavior**
I expected it to generate the SVG drawing of the hook.

**Screenshots or Drawings**
all i got was the pop-up with the traceback ..

**Additional context**
Add any other context about the problem here.

florianfesti, 2023-05-02:
Looks like the first line of the error message might be the reason for the failure. But this is kinda strange as you said it worked before. May be you changed something in your setup. The `boxes` executable needs to be in the path for this to work.

florianfesti, 2023-05-30:
This looks like a local problem and there is nothing that can be done here without further details. Closing.

Feel free to reopen if there are any new facts.

closed 2023-05-30 by florianfesti

alvarobyrne, 2023-11-13:
Hello @bitbasher and @florianfesti, 

above all, @florianfesti,   thank you for your dedication to maintaining this valuable tool.

I'm experiencing exactly the same issue as reported by @bitbasher. However, my setup differs in that I opted not to install Cairo and Homebrew. Here are the steps I took:

1. In hte cloned repo folder, Created a virtual environment: `python -m venv venv`
2. in the cloned repo folder, Installed requirements: `pip install -r requirements.txt`
3. I opted not to install cairo because I read at [boxes.py documentation » Using the Boxes.py API » Architecture](https://florianfesti.github.io/boxes/html/api_architecture.html#back-end) that

> Boxes.py used to use cairo as graphics library. It now uses its own - pure Python - back end. It is not fully encapsulated within the drawing methods of the Boxes class. Although this is the long term goal. Boxes.ctx is the context all drawing is made on.

4. Ran the setup script: `./setup.py install`
<details>
  <summary><strong>Log of running `./setup.py install`</strong></summary>

```bash
./setup.py install
running install
/Users/<USER>/.pyenv/versions/3.10.8/lib/python3.10/site-packages/setuptools/command/install.py:34: 
SetuptoolsDeprecationWarning: setup.py install is deprecated. 
Use build and pip and other standards-based tools.

  warnings.warn(
/Users/<USER>/.pyenv/versions/3.10.8/lib/python3.10/site-packages/setuptools/command/easy_install.py:144: 
EasyInstallDeprecationWarning: easy_install command is deprecated. 
Use build and pip and other standards-based tools.

  warnings.warn(
running bdist_egg

running egg_info
writing boxes.egg-info/PKG-INFO
writing dependency_links to boxes.egg-info/dependency_links.txt
writing requirements to boxes.egg-info/requires.txt
writing top-level names to boxes.egg-info/top_level.txt
reading manifest file 'boxes.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE.txt'
writing manifest file 'boxes.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-13.1-x86_64/egg

running install_lib
running build_py

updatePOT()

generate_mo_files()

buildInkscapeExt()
creating build
creating build/lib
creating build/lib/boxes
copying boxes/extents.py -> build/lib/boxes
copying boxes/vectors.py -> build/lib/boxes
copying boxes/walledges.py -> build/lib/boxes
copying boxes/Color.py -> build/lib/boxes
copying boxes/gears.py -> build/lib/boxes
copying boxes/drawing.py -> build/lib/boxes
copying boxes/__init__.py -> build/lib/boxes
copying boxes/qrcode_factory.py -> build/lib/boxes
copying boxes/edges.py -> build/lib/boxes
copying boxes/svgutil.py -> build/lib/boxes
copying boxes/formats.py -> build/lib/boxes
copying boxes/parts.py -> build/lib/boxes
copying boxes/robot.py -> build/lib/boxes
copying boxes/pulley.py -> build/lib/boxes
copying boxes/lids.py -> build/lib/boxes
copying boxes/servos.py -> build/lib/boxes

[... 698 lines cut ...]
Adding affine 2.4.0 to easy-install.pth file

Using /Users/<USER>/.pyenv/versions/3.10.8/lib/python3.10/site-packages
Searching for numpy==1.25.0
Best match: numpy 1.25.0
Adding numpy 1.25.0 to easy-install.pth file
Installing f2py script to /Users/<USER>/.pyenv/versions/3.10.8/bin
Installing f2py3 script to /Users/<USER>/.pyenv/versions/3.10.8/bin
Installing f2py3.10 script to /Users/<USER>/.pyenv/versions/3.10.8/bin

Using /Users/<USER>/.pyenv/versions/3.10.8/lib/python3.10/site-packages
Finished processing dependencies for boxes==0.9
```

</details>

After that, the `boxes` command is executable in my terminal after installation, indicating that it's in the system's PATH. (Maybe **inkscape** cannot access it? but why?)

5 . Looked into inkscape, `settings >> system >> user extensions`, to see the folder, which is `/Users/<User>/Library/Application Support/org.inkscape.Inkscape/config/inkscape/extensions` and copied the files generated by the install command in the `inkex` folder, there

@bitbasher, did you find a resolution to this issue? I'd appreciate any insights you can share. 

And @florianfesti, do you have any suggestions or known fixes for this situation?

Thank you for your assistance.

florianfesti, 2023-11-13:
You might need to also copy `scripts/boxes_proxy.py` to the `inkex` folder.

alvarobyrne, 2023-11-15:
Thank you for the suggestion, @florianfesti. Checking the function call stack in the error message, seems that the `boxes_proxy.py` file is indeed being invoked during the process.

What is not clear to me is the specific error message I'm encountering related to an XMLSyntaxError indicating an empty document:

```
/bin/sh: boxes: command not found
Traceback (most recent call last):
  File "/Users/tutree/Library/Application Support/org.inkscape.Inkscape/config/inkscape/extensions/boxes_proxy.py", line 82, in <module>
    main()
  ...
lxml.etree.XMLSyntaxError: Document is empty, line 1, column 1
```

Given this error, it appears that the issue might not be directly related to the invocation of `boxes_proxy.py`. Instead, it seems to stem from a problem parsing an empty document.

Do you have any further suggestions on how to troubleshoot this specific `XMLSyntaxError` related to an empty document in the context of the `boxes.py` extension?

Your guidance would be greatly appreciated.