florianfesti/boxes #557: TrayType Sides Too Long

scottdidit, 2023-05-03:
**Description**
I create a tray type with 3 internal sections; 2 side-by-side on one end, then 1 section taking the rest of the box.  The sides, as well as the dividers are too long.  In my case, about 1.2mm too long.

**To Reproduce**
I don't have an URL, but I attached the settings I used (.pdf file).  I also attached the cut file (.svg) which contains notes pointing out the problem.

**Expected behavior**
Sides and dividers should be the correct length for the bottom.

**Screenshots or Drawings**
-none-

**Additional context**
Add any other context about the problem here.
[SampleBox02.pdf](https://github.com/florianfesti/boxes/files/11379114/SampleBox02.pdf)
[image]

florianfesti, 2023-05-03:
Yeah, turns out the base plate doesn't take the `extra_length` in the `FingerJoint Settings` into account. I wonder where else the code still assumes that the finger joints are exactly one thickness deep. Luckily the default corner does take the actual value into account. This is why the walls are bigger (as they should be). Anyway, should be reasonably easy to fix for the `TrayLayout` at least.

scottdidit, 2023-05-12:
Thank you for responding, sorry my response back was a little slow.

How will I know when the fix has been deployed to the on-line generator page?

-=|Scott|=-

florianfesti, 2023-05-14:
I will update this issue as soon as there are news.

I actually already started working on this, but the code for the base plate is a bit intricate and needs new parts to handle the corners. There's currently a lot of other stuff going on, so it may take a while longer.

closed 2023-05-16 by florianfesti

florianfesti, 2023-05-16:
Change just got added to the server. Please test and re-open this ticket if there are still issues.

scottdidit, 2023-05-17:
Thank you so much.

-=|Scott|=-

commit aadae072 (2023-05-16): TrayLayout: Take true width of edges into account

for the base plate. Don't assume a width of one thickness as this fails
if extra_length is used in the FingerJoint Settings.

Resolves: #557