florianfesti/boxes #558: Generated SVG file does not contain units

rshouker, 2023-05-06:
**Describe the bug**
Generated SVG files does not contain units, and therefore all dimensions treated as pixels.

**To Reproduce**
[Simple Closed Box.](https://www.festi.info/boxes.py/ClosedBox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&x=140&y=75&h=15&outside=0&thickness=3.0&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&reference=100&inner_corners=loop&burn=0&language=en&render=0)

**Expected behavior**
When importing the file to CAD / laser cutting software it requires scaling

**Screenshots or Drawings**
[image]

**Additional context**
I fixed it by surrounding all groups with a group with scale attribute. 
```<g transform="scale (3.7795, 3.7795)">```

florianfesti, 2023-05-16:
Well, the document size is actually given in `mm` as can be seen in the second line:

```
<?xml version='1.0' encoding='utf-8'?>
<svg width="313.90mm" height="157.50mm" viewBox="0.0 0.0 313.90 157.50" xmlns="http://www.w3.org/2000/svg" ...
```
So it looks like your software is not honouring that. Is there any meta data that could be added to help with that?

florianfesti, 2023-05-30:
Closing as not a bug. Feel free to reopen and attach the header of and SVG that shows up in the right scale.

closed 2023-05-30 by florianfesti