florianfesti/boxes #561: How to use cartesian coordinates

ccrome, 2023-05-25:
Hi Florian,
   I'm curious if it's possible to use cartesian coordinates for drawing things.   I have a bunch of coordinates and just want to create cutouts and vectors at exactly those locations, without any fancy corner or offset calculations.

florianfesti, 2023-05-26:
You can use the ctx methods directly. See drawpoints() in https://github.com/florianfesti/boxes/blob/master/boxes/gears.py#L388 

The issue with that is that it doesn't do any kerf correction. So there is kerf in https://github.com/florianfesti/boxes/blob/master/boxes/vectors.py#L94 to help with that.

May be drawpoints() should be moved to the main Boxes class to allow this kind of thing more easily. Patches welcome.

ccrome, 2023-05-26:
Cool.  Looks easy enough to move.  I'll move it over and open an PR.

ccrome, 2023-05-26:
By the way, it looks like the existing drawPoints already did do kerf correction.

florianfesti, 2023-05-26:
Yeah, drawPoint does kerf correction but the ctx methods themselves don't.

Guess this is solved now.

closed 2023-05-26 by florianfesti