florianfesti/boxes #567: [Request] Inverse FingerHolesAt

deepinthewoods, 2023-06-22:
It would be useful to have fingerHoles that fit the other (F) edge type.

florianfesti, 2023-06-22:
Well, we had that discussion already around the addition of HandleEdge. The issue is that f edges are not guaranteed to have any fingers. As a result this new edge could be all hole splitting the part in two.

See `handle` parameter in typetray as one example on how we avoided the issue by switching to an `f` edge instead of adding a new finger hole edge type.

closed 2023-06-22 by florianfesti

deepinthewoods, 2023-06-22:
Okay, but was just talking about the fingerHolesAt function for making finger holes in the middle of a piece i.e I have recreated [this](https://dadamachines.com/wp-content/uploads/2018/02/dadamachines-automat-toolkit-mallet_0001.jpg), but I can't do the H shaped bridge section between the pieces...