florianfesti/boxes #574: Inkscape Plugins disabled

rebeltaz, 2023-08-30:
I installed the inkscape plugins using the manual method under Ubuntu 22.04 in inkscape 1.1. The plugins shows up under the extension menu, but every one is greyed out / disabled. What am  I doing wrong?

rebeltaz, 2023-08-31:
Further research has led me to believe that this extension is not compatible with inkscape v1.0

https://inkscape.gitlab.io/extensions/documentation/authors/update1.0.html

...
Remove dependency listings

Remove the dependency listings for the following modules:

    bezmisc.py

    coloreffect.py

    cspsubdiv.py

    cubicsuperpath.py

    ffgeom.py

    inkex.py (removal not strictly required)

    pathmodifier.py

    simplepath.py

    simplestyle.py

    simpletransform.py

    more?
...

as these are the same files being complained about.

rebeltaz, 2023-08-31:
if anyone has this same trouble, you can run the following commands from the 'boxes-master/' directory

```
sed -i 's/env python3/python3/' scripts/*
setup.py
scripts/boxes2inkscape
mkdir ~/.config/inkscape/extensions/images 
cp static/samples/* ~/.config/inkscape/extensions/images
cd ~/.config/inkscape/extensions/
sed -i "s/<image>\./<image>images/" boxes.py.*.inx 
```

abergmeier, 2024-03-07:
Isn't the problem with `python3` rather the Windows users, that commit CRLF?

Deltaplc, 2024-03-16:
Good morning
Excuse my ignorance, but I haven't yet understood the procedure to create all the files necessary to use this beautiful application on inkscape!!
How do you create the boxes.exe file? And all the inx files?
Can you help me?
Can you give me a step by step guide?
I can't really understand from the instructions on the site!!!
Thank you

florianfesti, 2024-08-05:
You might want to try MightyScape a selection of pre-build Inkscape extensions including Boxes.py:  https://gitea.fablabchemnitz.de/FabLab_Chemnitz/mightyscape-1.2-zipmirror

closed 2024-08-05 by florianfesti