florianfesti/boxes #575: QR code generator in inkscape plugin  dos not show numbers

jgforbes, 2023-08-31:
**Describe the bug**
QR code generator in inkscape plugin  dos not show numbers only parameter names.

**To Reproduce**
Fedora 38, Python 3.11 inkscape 1.3 and latest boxes.

**Expected behavior**
QR code with values for parametes

florianfesti, 2023-08-31:
Can you, please provide a link? Note that the QR code will only include parameters that are different to the default values. If you hit `Generate QR code` without changing anything you will basically get a link without any parameters. Try changing the sizes to something else than the default to get those added to the QR code.

While this behavior makes the generated QR code dependent on the defaults it makes them much smaller and more robust to changes - especially within the edge settings.

florianfesti, 2023-12-17:
Closing due to no activity. Feel free to re-open if issue still exists.

closed 2023-12-17 by florianfesti