florianfesti/boxes #588: GridfinityTrayLayout: add horizontal 'shelf' feature for making cutouts for specific items

ccrome, 2023-10-01:
I want the ability to automatically create horizontal, full coverage shelves in the gridfinity layout.  This will let me easily make cutouts for screwdrives, and other custom items

**Describe the solution you'd like**
Add another 'floor', and floor cutouts at specified heights.  Valid only when countx==county==1

Please assign this to me. I will update.

ccrome, 2023-10-01:
What's the easiest way to proceed with adding a 'shelf' in these 1x1 gridfinity boxes.  Basically what I think I'm looking for is this:
[image]

florianfesti, 2023-12-17:
It's probably easier to take the DrillBox and turn that into a GridFinityDrillBox generator. The issue with the TrayLayout is that there are multiple sections and you can only anker horizontal wall with finger joint from one side.

closed 2024-08-01 by florianfesti

ccrome, 2024-08-20:
ooh, this is great! just cut myself one.   images incoming.

commit b6934155 (2024-08-01): New generator: GridfinityDrillBox

Gridfinity box to store drills, router bits, etc

Currently doesn't support generating the hole pattern.

Resolves: #588