florianfesti/boxes #602: Ability to execute custom generator

sethfischer, 2023-11-07:
## Proposal

Add the ability to execute a [custom generator](https://florianfesti.github.io/boxes/html/api_generator.html).

## Use case

I keep a record of commands used by maintaining an [invoke](https://www.pyinvoke.org/) `tasks.py` file. Others may maintain a `Makefile` to keep a record of their boxes.py builds.

In either case, boxes.py would be installed into a virtual environment using [pip](https://pip.pypa.io/en/stable/), [Poetry](https://python-poetry.org/), or any other Python package manager (see example below). This has the advantage of pinning a specific version of boxes.py and dependencies to ensure reproducible output.

When installed in this manner is it not convenient to add custom generators alongside the bundled generators (which are in a managed virtual environment).

I would like to add [custom generators](https://florianfesti.github.io/boxes/html/api_generator.html) to this project, say in a `my_generators/` directory.

## Proposed solution

Add the ability to execute custom generators from the `boxes` executable script, for example:

```
boxes MyGenerator --generators-dir="path/to/my_generators/" [boxes default arguments]... [my generator arguments]...
```

or similar.

## Example

An example Poetry `[tool.poetry.dependencies]` table from [sethfischer/boxes-builds](https://github.com/sethfischer/boxes-builds):

```toml
[tool.poetry.dependencies]
python = "^3.9"
boxes = {git = "https://github.com/florianfesti/boxes.git", rev = "2ab7fc909eec106af17bad9347289c07e626277c"}
invoke = "^2.0.0"
```

shish, 2023-11-26:
A thing that's working for me is making the generators themselves executable:

```python
#!/usr/bin/env python3

from boxes import *

class MyCustomThing(Boxes):

   ...

if __name__ == "__main__":
    box = MyCustomThing()
    args = """
        --output=teabox-3mm.svg --format=svg
        --thickness=3 --burn=0 --reference=0 --outside=false
        --sx=366 --y=300 --h=126 --front_height=0.7
    """.split()
    box.parseArgs(args)  # or sys.argv[1:]
    box.open()
    box.render()
    box.close()
```

This could probably be a bit more elegant (eg if `class Boxes()` had a convenient method to combine parseArgs + open + render + close) but it's working for now

florianfesti, 2024-01-27:
Using a env variable seemed to be the easiest way to get this done quickly. Please test!

closed 2024-01-27 by florianfesti

sethfischer, 2024-02-02:
@florianfesti An elegant solution - thank you!

commit 3369dfe0 (2024-01-27): Add support for BOXES_GENERATOR_PATH

Allow reading generators from a separate location.

Resolves: #602