florianfesti/boxes #610: Boxes and Blender Feature. List arguments from a generator

Franckapik, 2023-12-20:
Hi,

First of all, thank you a lot for this library i use since few years.
I am working on integrate the library in a blender addon to generate boxes directly inside the 3D workspace. 

I am trying to use boxes API but i am wondering how is it possible to get the list of args for a generator ?
Is there a line in the package which give an example of function listing arguments to use them next with inputs ?

Thank you for your help !
F.

florianfesti, 2023-12-20:
The arguments for each generator are stored in the `argparser` attribute which is a `argparse.ArgumentParser` object. You can get the list on the command line with `./boxes GENERATORNAME --help`. Have a look at `scripts/boxes2inkscape` for how to extract the parameters in a python script. That probabyl your best best to convert the parameters into what ever format Blender needs.

florianfesti, 2023-12-31:
I hope this answers your question. Closing for now to not keep this open for ever as there is no further actions to be done upstream. Feel free to continue with questions here if you run into problems.

closed 2023-12-31 by florianfesti

Franckapik, 2024-01-02:
Thank you a lot for your response ! I have worked on it and the have the addon to have integration of Boxes in Blender !

[image]

[image]

I know that people should be interesting by this project, for example people using [blenderCAM](https://www.blendercam.com) .

@florianfesti , What could be the best way to offer to others this new addon ? Pull request on your repo or distinct repo with maybe a link in documentation ? 

Thank you !!

florianfesti, 2024-01-02:
A pull request in the main repo is probably the best way. Having things separate only leads to stuff getting out of sync.

The interesting question is how to offer an installable binary package. That's a topic I have not really put in the necessary work. This is not at all limited to your new Blender plugin but also for Boxes.py itself and the Inkscape plugin as well.

So if you have an idea on how to solve that in a general way that would be fantastic. But it's also fine to have a Blender only solution (a least at first). Building and offering it here on GitHub would be preferred, but if you want to build and upload it on your own page or a general Blender plugin page that's also fine with me.

Franckapik, 2024-01-02:
Ok! I agree with you !
I should try to make a pull request first ! I fork your repo and add a "blenderboxes" folder witch is the blender addon that we can import inside blender after cloning your repo on local machine. 
You can see the added files here : https://github.com/Franckapik/boxes/tree/master/blenderboxes
I probably could add the new integration in the documentation too by modifying the files in your repo? 

Thank you !

Franckapik, 2024-01-02:
The question is, should i have to make pull request then each time i update the code on my fork ?

Franckapik, 2024-01-03:
I have read about multi-platform installer (like Qt ), it could be interesting to work on a binary for windows and linux.
Have you think about upload the boxes library in Pypi to install it directly with pip ? 
It could be the best way on the side of blender to install and import boxes in the blender addon.

florianfesti, 2024-01-07:
PyPi is probably the way to go. Unfortunately I have not found a good generic way to package non trivial Inkscape plugins. No idea how Blender plugins are typically packaged.

One challenge is that different tools can use different Python interpreter. While the Linux approach tends towards using the system installation other distributions or OSs may bundle their own Python with the applications. Fun for the whole family...