florianfesti/boxes #611: Corners missing on TrayLayout

Zenock, 2023-12-29:
**Describe the bug**
I created a box here https://www.festi.info/boxes.py/TrayLayout?language=en using TrayLayout. When I export the box as an svg the corners are missing on the base of the box/tray.  All the other pieces have all the corners and the tabs have all their corners only the 4 corners of the base are missing.

**To Reproduce**
https://www.festi.info/boxes.py/TrayLayout?language=en
The settings I used were:
h=140 
hi=0
outside unchecked
sx=50*2
sy=140

 ,> 50.00mm
 | ,> 29.50mm
+-+-+
| | |  140.00mm
+-+-+
Thickness=3.0
format=svg
Tabs=0.0
qrcode, debug, and label unchecked
reference=0
inner_corners=corner
burn=0.1

When I do this, if I zoom in real close to the 4 corners of the base and only the base piece, they are missing the curved part of the corner leaving small gaps.  The gap is 0.1 mm in the x and y directions leaving an overall direct gap of the sqrt of 0.2. I can and have draw this curve in by hand but it is annoying and it ONLY happens on the base piece.  

**Expected behavior**
I would expect the base piece to be a closed path.  Not sure what else to say here.

**Screenshots or Drawings**
[image]
[image]

**Additional context**
I really enjoy this and really shouldn't complain.  Maybe I'm doing something wrong?  It's not so terrible I can't deal with it.  Just thought it might be worth pointing out.

maihde, 2025-05-07:
I took a look at this - the issue is that the base generator wasn't drawing corners/taking into account the burn value.  I made a few minor changes and at least for my test cases is now rendering proper corners on the base.

[image]

florianfesti, 2026-01-18:
OK, this is fixed. Inner corner probably should also get proper corners but that's not quite as urgent as the lines just overlap no  matter what.

closed 2026-01-18 by florianfesti