florianfesti/boxes #615: Air Purifier Box Generator- will pay

MachineryEnchantress, 2024-01-10:
There is a small community of people who DIY very quiet, very powerful HEPA filters using PC fans- mostly for classrooms. I have created several templates for them: https://github.com/opennukit/Nukit-Open-Air-Purifier/

Users need to be able to design their own boxes by specifying filter dimensions, how many filters, the number of fans, on which sides of the box, and if they are 120mm or 140mm fans.

My Python is a little rusty but I'm happy to pay, or barter. I can help you get pretty much any hardware sourced, fabricated or manufactured in Shenzhen that you might ever want.

Thanks to anyone interested🙇🏻‍♀️

[image]

florianfesti, 2024-01-10:
This is just a simple box. I cobbled something together: https://festi.info/boxes.py/AirPurifier
Still misses a few details like screw holes for the fans. I also wonder if we really need those large panels which are basically all hole.  May be we can get away with just rails left and right?

MachineryEnchantress, 2024-01-11:
Thank you! The filters are made of paper and polypropylene- so not very stiff, if they aren't pressed a bit on all four sides the air purifier will leak and the CADR drops by a lot. The big empty square is easy to cut on half later.

This is what I had it mind:

The filter slots on each side are mirrored and the filter XYZ size is defined by three variables in mm. This, along with the specified fan size (140mm or 120mm), determines the overall dimensions of the box.

So we need to be able to define the size of the PC fans used. 
One variable- 120mm/140mm

The inner channel will always be one of two fixed widths defined by which size fans were selected above- 142mm wide for 140mm fans, or 122mm wide for 120mm fans.

The user can select one, left or right, or two filters. In the case that one filter is set, the inner filter lip on one side of the box becomes closed (with finger joints). 
So one variable for number of filters- two, one left, one right

We need to be able to define the number of those PC fans on each side (front, back, top, bottom since sometimes these are hung from the ceiling horizontally). These are arranged in a single row. 
Four variables, 0-12.

No less then 27mm between fan mounting screw hole centers, but otherwise evenly distributed.

Main fan hole edge no closer than 30mm to box edge.

The height of the filter lips are a single variable in mm.

Material thickness is a single variable in mm.

Finger length is a single variable in mm.

Fingers or no is a single variable Y/N.

It’s a lot😣 That why I’m happy to pay or offer something.

florianfesti, 2024-01-11:
Nah, that easily fits in just 140 lines. I think I got everything for now. May be test cut a small model and check if everything fits together.

MachineryEnchantress, 2024-01-12:
Amazing! Perfect on the first go💪🏻 I just dry fitted it for testing, will glue one up for a picture. (Ignore the bulge in the back there, that's me, I didn't leave enough kerf when I split the part and it's a bit tight😅)
[image]

The original has a maximum of three fans per side, this maxes at two. Can we make it three? With only four in total the CADR will be quite low...

Huge ask, if it’s a pain don’t bother, is it possible to have cutting the big filter frame pieces in two as a built in option? A lot of the people making these are parents on library Makerspace 6040 CO2 cutters that will probably get stuck at trying to split the part (or mess it up like I did the first try). But if it’s a pain never mind.
[image]

These settings as the default work perfectly with a standard HVAC MERV 13 filter:
https://festi.info/boxes.py/AirPurifier?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&x=498.0&y=496.0&filter_height=46.77&rim=30.0&fan_diameter=140&filters=2&fans_left=-1&fans_right=-1&fans_top=0&fans_bottom=0&screw_holes=5.0&thickness=3.0&format=svg&tabs=0.0&qr_code=0&debug=0&labels=0&labels=1&reference=100&inner_corners=loop&burn=0.1&language=en&render=0
[image]

Thank you so so so much- please let me know what I can do in return- now or anytime🙏🏻

florianfesti, 2024-01-12:
OK; three minor tweaks:

* Default settings matching the ones above
* Squeezed out a bit of space between the fans to make 3 140mm fans fit.
* Make the top (on the photo) flush by using extended edges and making the inside on thickness lower

I think instead of cutting the pieces holding the filters in two I will divide them up in 4 rectangles. With dovetail joints that should be robust enough. We can still add stiffeners later if they are not sturdy enough. May take another day or two, though.

florianfesti, 2024-01-13:
A little bit of insomnia goes a long way...

Here's the new `split_frames` parameter that turns the frame pieces in to four separate sides connected by dove tails. Not only will this make all pieces fit on a 60x40cm machine but now the 2 filter variant will need the same or slightly less material than the one filter one.

With so many changes we probably need a few more test cuts to cover at least some of the variants. Assembly instructions with pictures would also be nice. This is now a pretty nice 3D jigsaw puzzle...

PS: https://festi.info/boxes.py/JointPanel offers a better method to split large piece and joining them back together

MachineryEnchantress, 2024-01-13:
Fantastic, will test and let you know- thank you!

MachineryEnchantress, 2024-01-14:
I *think* this is a misslice? No? Default box.
[image]
[image]

florianfesti, 2024-01-14:
No, this is intended. These are two separate Finger Joint Hole sections. One for the cross part and one for the side piece of the frame. If you assemble the frame first you should have something that just fits into there.

MachineryEnchantress, 2024-01-14:
Ok will double check...

MachineryEnchantress, 2024-01-14:
The top makes sense, doubled tabs on the side panel, continuous on the back:
[image]

...but still working on this, doubled tabs on the back **and** the side panel:
[image]
[image]

Will let you know if I figure it out😅

florianfesti, 2024-01-14:
Oops, indeed. I am currently AFK. Will fix in a few hours Von meinem/meiner Galaxy gesendet

20240114_170546.jpg (view on web)
20240114_170409.jpg (view on web)
20240114_170322.jpg (view on web)

MachineryEnchantress, 2024-01-14:
Oh thank goodness I thought I was going crazy😅 No rush at all, I'm immensely grateful for all this.

florianfesti, 2024-01-14:
OK, should be fixed with c40cb4249af530c08e4e9d013f13dd37899bec5b. No worries. I don't feel rushed. I am just slightly annoyed at myself that I didn't quite get what the issue was right away. This is a super simple thing to fix and I easily could have done that thi morning before leaving the house. Otoh this was all before my second coffee - so I don't take any responsibility for that ;)

MachineryEnchantress, 2024-01-16:
Fantastic! It works perfectly! Thank you so much Florian! If there's ever anything you need from Shenzhen- you have my email, it's yours!
[image]
[image]

florianfesti, 2024-01-18:
As I pointed out above Boxes.py has [Joint Panel](https://festi.info/boxes.py/JointPanel) to divide larger pieces. Small Dovetail joints (Adjust `depth` and `size` in the DoveTail Settings) with tight joints can be hammered together to form basically one pieces that is just as strong.

closed 2024-01-25 by florianfesti

florianfesti, 2024-01-25:
OK, I declared the generator as complete. If anyone has better images or assembly instructions feel free to open an PR or just post them here.

commit 2a055f89 (2024-01-25): AirPurifier: Declare as stable

Add picture

Resolves: #615