florianfesti/boxes #623: Local installation error: No module named 'markdown'

D-Platoon, 2024-02-09:
_Originally posted by @florianfesti in https://github.com/florianfesti/boxes/issues/195#issuecomment-564739556_

I have the same problem as the user in the issue above while installing boxes.py to my local computer (Windows 10).
I installed Python 3.12.2 (latest version), and i think i have done every other steps like in the description.
At Step7 I will also get the following error:

D:\boxes-master>python scripts\boxesserver
Traceback (most recent call last):
  File "D:\boxes-master\scripts\boxesserver", line 34, in <module>
    import markdown
ModuleNotFoundError: No module named 'markdown'

If I want to reinstall markdown i get the following message:

D:\boxes-master>pip3 install Markdown
Requirement already satisfied: Markdown in c:\users\[User]\appdata\local\programs\python\python312\lib\site-packages (3.5.2)

lxml is also already installed.

Any ideas what else I can do?

florianfesti, 2024-02-09:
This might be an issue of having both Python  2 and Python 3 installed. Try `python3 scripts\boxesserver`

D-Platoon, 2024-02-09:
Thank you for your fast advise.
But i have no Python 2 installed and with this command I get also an error:
`Python konnte nicht gefunden werden. F³hren Sie die Verkn³pfung ohne Argumente aus, um sie ³ber den Microsoft Store zu installieren, oder deaktivieren Sie diese Verkn³pfung unter`

Now I tried to install it on my virtual machine (also Win10) and there it works properly.
I don't understand, whats the problem with my local machine.
Do you have any other ideas?

florianfesti, 2024-02-09:
I don't have any experience with Python under Windows. I'd guess that `python` and `pip3` use different search paths. May be have a look if `python -c "import sys; print(sys.path)"` yields anything.

D-Platoon, 2024-02-09:
Big thank you - thats the solution!
There is Inkscape on my computer.
First time I run the command I get:
`['', 'C:\\Program Files\\Inkscape\\lib\\python310.zip', 'C:\\Program Files\\Inkscape\\lib\\python3.10', 'C:\\Program Files\\Inkscape\\lib\\python3.10\\lib-dynload', 'C:\\boxes-master', 'C:\\Program Files\\Inkscape\\lib\\python3.10\\site-packages']`

As I uninstalled Inkscape it has found the right path to python. And now I can execute boxes.py
Now i just need to have a look, how I can manage it to have also Inkscape installed.

I'm not good in python, but now I wan't to try some coding.
Add some finger holes to typetray (found it at cardbox) and I want to mark every tray with a text from a list (probably more difficult).
But maybe it's better to open a new issue for this, right?

florianfesti, 2024-02-09:
Well, unless you need a lot of those it is probably easier to re-install Inkscape and add both of those to the SVG.

But yes, such features deserve their own ticket.

florianfesti, 2024-02-09:
I guess the problem is that `python .. ` executes the Inkscape one. If you use the full path to the other Python instance you should not have that issue.

D-Platoon, 2024-02-09:
Jep, that make sense.
Tried it and with the full path it works also with Inkscape installed.
Thanks a lot!

closed 2024-02-09 by florianfesti