florianfesti/boxes #624: Web site loads slowly

chrysn, 2024-02-10:
**Describe the bug**

The website (which is just the repository's scripts/boxserver) often takes several seconds to load -- so long that when I show people how cool this tool is and how easy it is to use, I have enough time to look over to them and see the frowning faces while the page appears to be down.

**To Reproduce**

I couldn't reproduce this locally, but both a browser request to `https://festi.info/boxes.py/` and a `curl` currently take between 20 and 30 seconds.

**Expected behavior**

There are likely recommended values for page loads known to those active in web development, but something <1s would be practical.

**Screenshots or Drawings**

[image]

This shows that the time is spent during TLS setup, which may explain why it isn't showing locally.

florianfesti, 2024-02-10:
Yeah, I am aware of that. The web instance will move to a new home the next few weeks.

PaulKierstead, 2024-03-17:
I actually wonder if it is the instance or some other bug. I just stood up two instances using the current Dockerfile as of this comment. The first one I did on macOS using apple silicon. When I access it locally, it is extremely snappy all the time. So then I moved over to my ubuntu based x86. That one is snappy .... sometimes. If, for example, I just click back and forth between Gallery And Menu, *sometimes* either one will take over 40 seconds to respond. If you look at the network graph in the browser developers tools, the Gallery or Menu endpoint will have a very high latency, but the actual load time will be normal. So you'll see a latency of like 42 seconds, but duration of 8 ms. Last one just had 59 seconds latency.  Same browser in both cases, network is local and reasonably quick (and the problem is no response at all, not slow loading). Never seems to happen in the Mac OS instance.

PaulKierstead, 2024-03-17:
My apologies, I should have tried but it really didn't seem browser related, however ... if I use Firefox (instead of Safari) I get no delays on either instance, including from the same machine/network/etc that was getting delays using safari. Windows Firefox is also fine. I have no idea what could cause that.

Also, I'd love to add I love your open source project and just how easy it is to stand up an instance. A shining example of doing it right, and making things so easy, and great software.

PaulKierstead, 2024-03-18:
One final piece of weirdness: I put nginx as reverse proxy in front of the boxes container, and safari doesn't have any issues then. But if I go directly to the container, it still has issues. Not urgent, but very weird.

florianfesti, 2024-03-22:
This is all very weird. Especially as the Dockerfile in the repository is super simple - unlike the one I use on the server which uses uwsgi behind apache.

florianfesti, 2024-05-12:
So the instance is now running on the new server of my hackerspace: https://boxes.hackerspace-bamberg.de

I added a redirection on my website. I am going to test this a few days and then change over the links in the documentation, the hackaday.io project, ...

florianfesti, 2024-05-12:
Oh, please test and report anything weird or slow.

florianfesti, 2024-05-26:
Things seem to run smoothly. Closing this ticket as solved - even if we really still have no clue what the issue is/was.

closed 2024-05-26 by florianfesti