florianfesti/boxes #627: Document guidance on model influenced edge parameters

chrysn, 2024-02-12:
While most edge parameters are largely a matter of preference ("How much space between finger joints" doesn't change whether a model works), some parameters originate from the way the model is set up (for example, when a model uses two sheets on the floor, the "h - Edge (parallel Finger Joint Holes)" for the side walls would need an extra parameter that makes the finger holes twice as wide.

Right now, AIU, there is only one such set of parameters, that is the bed bolts -- and that is an even trickier case because while in some models it may be optional whether or not bed bolts are used, in some models they work only on specific edges (where they don't interfere with some part of a sliding mechanism). That bed bolts go both ways is illustrated by my proposal to extend their arguments to layers (in https://github.com/florianfesti/boxes/pull/625) and @xloem's proposal to make them just an argument option (in https://github.com/florianfesti/boxes/pull/468).

What is the general pattern that model dependent edge parameters should use? I'm leaning toward copying the bedBolt pattern of being a keyword argument to the edges that is passed around (and ignored by virtue of **kw). Is that the pattern to go with?

The concrete edge parameters I'd like to introduce are variations for dual-layer peers -- depending on how my current project goes, that may be just multiple layer thick parts (that'd then have double long fingers or double wide slots in "h" style edges, possibly alternating slots in "h" style edges with the corresponding parameters for the matching "f", or more complex variations of "h" that allow glueless operation).

(Title-wise this is phrased as a documentation wishlist item -- practically it's more of a "question" type issue, and maybe once it is resolved it'll just be input to upcoming PRs that then already do establish a pattern).

chrysn, 2024-02-12:
An even better example might be the "angle" setting. The angle between two parts is not something the user configures, but it is a property of the model (eg. ABox is always at 90° angles; in models such as AngledBox it's configurable but even then it's not constant across all joints). In that example, the settings are deepcopied and applied to a new set of edge objects. That works even with the letter short-cuts because the angles are constant. (But it'd even work for non-constant angles because instead of letters the actual objects can be passed).

Is that the generally recommended pattern? If so, should options that can't be practically set (or are always set by the model) be hidden from the web UI and the command line? Should there be a helper function for "take these settings but adjust some details and create a new edge type out of it"?

florianfesti, 2024-03-28:
OK, after thinking about this for a while I guess the right way is making this a property of the Settings class without using the parameter mechanics. So the angle should not be in `absolute_params` but a simple property . That way the user can meddle with it and one can still create a copy of the settings to create a new set of edges - or change it in the code.

florianfesti, 2024-08-02:
OK, fixed `FingerJointSettings.angle`.

commit c3d93932 (2024-08-02): Settings: Document use of non public properties

Resolves: #627
closed 2024-08-02 by florianfesti