florianfesti/boxes #628: Sample Images missing

florianfesti, 2024-02-13:
Many "Box" generators still need a sample image. Please donate one if you create something that is still missing a picture! See [Contributing Document](https://github.com/florianfesti/boxes/blob/master/CONTRIBUTING.rst#provide-photos-for-generators) for details or just add the picture in a comment below.

The following generators are still missing a picture:

- CarbonFilter
- BottleStack
- DinRailBox
- Dispenser
- Display
- FillTest
- FlexBox5
- FlexTest2 - could be rendered image
- Gears
- Hook
- MakitaPowerSupply
- NemaMount
- Planetary
- PoleHook
- Pulley
- Rack*Box
- RectangularWall
- SevenSegmentClock
- SideDoorHousing
- SkadisBoard
- StorageRack
- Tetris
- TriangleLamp
- WaivyKnob 
- WallEdges
- WallRack

There was an old ticket (#140) for this which got too long.

D-Platoon, 2024-02-22:
It was not asked, but I have a new picture for the ConcaveKnob (haven't seen the WaivyKnob).
Maybe you can use it.
[image]

littleroblet, 2024-02-27:
I have added some images on Pull Request https://github.com/florianfesti/boxes/pull/632

Flexbox5
SideDoorHousing
Tetris

_I just wanted to test a few different things on my laser cutter so I picked a couple without images so I could contribute the images._

revjmoney, 2024-08-15:
[image]

Sliding Box Generator.  I needed a container for the metric tap set that amazon sent across the country loose in a plastic baggie?  thankfully they arrived intact.  Laser cutter for the win, now they have a home!

ClaneSharke, 2025-03-04:
FlexBox:

[image]
[image]

Cemelinio, 2025-05-07:
I think there is lots of room for improvement in the awesome original generic 7-segment clock template:
- Twisting LEDs around on the walls makes it challenging to place them in the middle of a segment. Instead, might be better to create more distance between slots in the back so that you could place the LED strip on the back panel in a parallel circuit.
- There is no space for a controller. I created a sliding box and placed it in the back. Added a couple of holes in the back for wires.
- There is no space for buttons. I added 2 at the top.

Other than that, I am super happy with this project (well, aside for the thin vellum paper I had for light diffusion...).

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

 I hope the images help.

YevhenCRAFT, 2025-05-14:
NemaMount

[image]
[image]
[image]
[image]

YevhenCRAFT, 2025-05-14:
TriangleLamp

[image]
[image]
[image]

JolienF, 2025-07-26:
[image]
[image]
[image]
[image]
Here are the pics for the sphere generator.

MartinScharrer, 2025-10-29:
Here also two images for the ConcaveKnob which has a sample image already for the default three serrations.

The images show it with six serrations:
[image]
[image]

florianfesti, 2025-10-30:
Thanks! Added as 6a1e4df31c6380e2b66894a684ee2e20d7864857

MartinScharrer, 2025-11-01:
Gears: 

[image]

MartinScharrer, 2025-11-01:
BottleStack (single):

[image]

florianfesti, 2025-11-03:
> Gears:
> 

What happened to those arcs?

MartinScharrer, 2025-11-03:
> > Gears:
> 
> What happened to those arcs?

I was wondering myself after upload. I thought it is from the generator, but there they are round. Maybe they got changed in LightBurn by accident. 

I think I will redo the larger one and reupload.

MartinScharrer, 2025-11-03:
> > BottleStack (single):
> 
> Would you mind taking a picture with a stack of bottles on top?

Sure, I can do.

MartinScharrer, 2025-11-04:
> > BottleStack (single):
> 
> Would you mind taking a picture with a stack of bottles on top?

[image]
[image]

MartinScharrer, 2025-11-04:
> > Gears:
> 
> What happened to those arcs?

I can now confirm that this always happens when downloading as lbrn2 format. I had to go over SVG import to get it right. 

[image]

florianfesti, 2025-11-04:
BottleStack images added as 3eaf302d704d99fe54dc28ff3e50050d09346db9

AlbertoGinel, 2026-10-03:
[image] I made myself a mechanical keyboard

florianfesti, 2026-10-04:
#958 has some very nice pictures of just simple boxes with lids and DrillBox'es. Linking here so they don't get lost.