florianfesti/boxes #630: TypeTray FingerHoles + Markingfield

D-Platoon, 2024-02-14:
Hi!

I'm not good in coding python but I wan't to extend the TypeTray-generator with two additional features:

1.) FingerHoles
I have found the module in the CardBox-generator. I hope, i will make it work and I have changed some code.
But now I wan't to test it, but I didn't found the code who is responsible for the web-sites.
Can anyone help?

2.) Markingfield
I wan't to mark the fields automatically by filling a textfielt (every line is an new field). Hope its not a big thing, but I have no idea how to manage it. I think there sould be a loop who searches for every CR/LF and write it into a new variable. The content of each variable should be positioned to the corner of every tray-divider. Maybe there should be a choice-box for the vertical or horizontal dividers.
Has anyone some ideas how to do it?

Here's a screenshot how it should look like:
[image]

florianfesti, 2024-02-14:
To get the finger holes you have to replace the top edge with your own edge.
Basically replace the `"e"` [here](https://github.com/florianfesti/boxes/blob/master/boxes/generators/typetray.py#L145). You can steal or borrow the FingerHoleEdge from the `CardBox`

For the text you need to add a callback to the [rectangularWall call just below](https://github.com/florianfesti/boxes/blob/master/boxes/generators/typetray.py#L150). See https://florianfesti.github.io/boxes/html/api_parts.html#the-callback-parameter or most other rectangularWall calls in TypeTray. DrillBox has text placed but it does calculate the content.

D-Platoon, 2024-02-21:
Today I tried some coding for the fingerholes.
Now i have a code who works. But now it has always fingerholes.
I added the new choice-option "none" (Line 108). The return-value is 0. This should be the information to disable the function.
But i don't know in which file i have to edit.

BTW: Can we discuss in german here? Maybe its easier.

The actual code is appended in txt because it's not allowed *.py
[typetray.txt](https://github.com/florianfesti/boxes/files/14365022/typetray.txt)

D-Platoon, 2024-02-23:
Ok, I solved the problem with the function disabling. I added just a if and it works:
[typetray - with Fingerholes.txt](https://github.com/florianfesti/boxes/files/14388213/typetray.-.with.Fingerholes.txt)

Now I will continue with the texts...

D-Platoon, 2024-02-24:
Okay, a long evening later I have a solution for all.
I didn't cut it yet, but I think, it should work. I will try it in the next few days.

Maybe you wanna check the code and give me a feedback.
[typetray_Fingerholes_Text.txt](https://github.com/florianfesti/boxes/files/14391258/typetray_Fingerholes_Text.txt)

florianfesti, 2024-02-24:
OK, this looks pretty nice already. I have moved the text rendering into a callback method. This make the placing a bit more straight forward.

I am a bit confused about your handling of the layout. `char2string` seems pretty strange and unnecessary to me. Is there a particular reason you added that?

I am also wondering if there should be an option to add the finger holes to the outside wall, too.

florianfesti, 2024-02-24:
Here's the current diff https://github.com/florianfesti/boxes/compare/master...typetext

D-Platoon, 2024-02-26:
I'm a PLC an robot programmer. I understand the most things about the logic in python, but I'm not good in coding a pretty file in this language. I just made it work like my preferences.

The callback method looks fine.

I used `char2string` just to convert a list of characters into a string. I tried to use the join-method but i couldn't get it to work. If you have a better idea, feel free.

Of course, I can add the option to add the finger-holes to the outside wall.

D-Platoon, 2024-02-26:
Okay, I worked on it.
The `char2string` is removed , now it's a join-method (see the code below)

The option to add the finger holes to the bront and back wall is not that easy I thought. I will get only one hole per wall, because the top edge is just one line (not like at the inner walls who are cutted from the vertical dividers).
Do you have an idea for that?

Heres my diff: https://github.dev/D-Platoon/boxes/blob/9520098d1dfd851b21e1a9b2695393af32214c47/boxes/generators/typetray.py

florianfesti, 2024-02-27:
OK, made FingerHoleEdgeSettings into a proper setting and use it in the UI. I also changed the defaults to have a the same behaviour as before. So both finger holes and labels need to be switched on.

I may squash most of the patches into just a few. May be tomorrow and then just merge.

closed 2024-02-28 by florianfesti

D-Platoon, 2024-03-02:
I wanted to try your upgrade. It looks pretty good, but I found a bug at the section of the compartment labels.
The `label_text` is just a line, but it should be a field. Because with the line, we can't get a differentation.
It looks like this:
[image]

But it should look like this:
[image]

I played a little bit with the code, and I found out, that if the variable is called `layout`, we're getting the field, otherwise we get just the line. But I don't know why...

I also think it is more useful to set the finger-holes by default with an absolute with of maybe 20mm instead of the relative widh (the thickness of your fingers is always the same - its independent from the compartment width).
And can you explain the variable `radius`? I tried different values but i got always the same output.

And I want to add some german translations. How can I do that?
E.g. "Forderseite" is written wrong ;-)

florianfesti, 2024-03-02:
Weird. The label_text works for me both locally and on the web instance. May be you are missing some patches. cc78633e930279a3b2ed709189c3db1d5df2c613 most likely.

The radius is meant to be the radius at the bottom of the finger holes. Turns out it wasn't hooked up top anything, though. Fixed with ad1010439ad83a6a4fcbed819f73e355694b30b4

florianfesti, 2024-03-02:
The translation source files can be found in `po/`

D-Platoon, 2024-03-02:
Okay, i didn't made a complete clone of the repository. My fault, sorry!

Also now the radius value make sense.

I think, I will test and cut today a small tray.

Thanks for the location of the language files. Think, I will update it in the next few days.

florianfesti, 2024-03-04:
While it's non of my business and really not prerequisite for contributing to Boxes.py, I still strongly suggest to learn git. It has changed the way I look at coding a lot to the better and having an easy way to rebase or merge branches is invaluable.

commit 13f36fd8 (2024-02-28): TypeTray: Add options for finger holes and compartment labels

Add FingerHoleEdge with setting class and an UI group for the labels

Resolves: #630
commit 55616e22 (2024-02-28): TypeTray: Add options for finger holes and compartment labels

Add FingerHoleEdge with setting class and an UI group for the labels

Resolves: #630