florianfesti/boxes #637: DXF Gernate failure

BHSDT, 2024-03-01:
**Describe the bug**
When entering box sizes on A Box, the DXF option fails to generate and produces an error message in a new tab.  It works in SVG format, so the issue is with the DXF generator.

**To Reproduce**
[If possible give a URL or command line that leads to the issue. If possible try to reproduce the issue on the public web instance.](https://festi.info/boxes.py/ABox?language=en)

Options selected:
114x205x300 internal dimension
Straight edge base
6.0 thickness
dxf
Tabs 0
No labels
reference 0
Inner corners loop
burn 0.1

**Expected behavior**
Normally this produces a DXF download immediately,  Now producing an error.

**Screenshots or Drawings**
Error message:
[If the bug can't be reproduced easily, add screenshots or attach a drawing to help explain your problem.](https://festi.info/boxes.py/ABox?FingerJoint_angle=90.0&FingerJoint_style=rectangular&FingerJoint_surroundingspaces=2.0&FingerJoint_bottom_lip=0.0&FingerJoint_edge_width=1.0&FingerJoint_extra_length=0.0&FingerJoint_finger=2.0&FingerJoint_play=0.0&FingerJoint_space=2.0&FingerJoint_width=1.0&Lid_handle=none&Lid_style=none&Lid_handle_height=8.0&Lid_height=4.0&Lid_play=0.1&x=205&y=114&h=300&outside=0&bottom_edge=e&thickness=3.0&format=dxf&tabs=0.0&qr_code=0&debug=0&labels=0&reference=0&inner_corners=loop&burn=0.1&language=en&render=2)

**Additional context**
Please sort this as a priority - I have several GCSE pupils using this for exam work that are now stuck...

florianfesti, 2024-03-02:
Easily reproduced with the web instance. It works here locally with a newer version of ps2edit. Guess I need to update the container to a newer release of the distribution. If you are in a hurry you can try to run Boxes.py locally for now.

BHSDT, 2024-03-02:
Hi Florian

Thank you so much for a personal reply - much appreciated.

As a non-programmer, how do I run Boxes.py locally?  What do I need to do?

Many thanks

Colin

MrCRichardsBSc, Cert.Ed.
Director of Studies, Head of Design & Technology
01625 422192
Beech Hall School is registered with the Department for Education as an authorised independent school. Registration number: 895/6008
Beech Hall School is part of Chatsworth Schools Ltd. Company Number 09540576 
Registered Office: Great Tew Estate Chipping Norton Oxfordshire OX7 4AH
This email and any attachments to it may be confidential and are intended solely for the use of the individual to whom it is addressed.
​Any views or opinions expressed are solely those of the author and do not necessarily represent those of Beech Hall School or those of Chatsworth Schools. 
​If you are not the intended recipient of this email you must neither take any action based upon its contents nor copy or show it to anyone.
​Please contact the sender if you believe you have received this email in error.

florianfesti, 2024-03-02:
There's https://florianfesti.github.io/boxes/html/install.html but I am not sure if it is that helpful for a Newbie.

florianfesti, 2024-03-03:
Oh, of course you can also just generate a Postscript file and convert with pstoedit yourself or create a SVG and use Inkscape to convert to DXF.

BHSDT, 2024-03-04:
Thank you - I assume converting the output file like this will retain the
dimensional accuracy?

Colin

MrCRichardsBSc, Cert.Ed.
Director of Studies, Head of Design & Technology
01625 422192
Beech Hall School is registered with the Department for Education as an authorised independent school. Registration number: 895/6008
Beech Hall School is part of Chatsworth Schools Ltd. Company Number 09540576 
Registered Office: Great Tew Estate Chipping Norton Oxfordshire OX7 4AH
This email and any attachments to it may be confidential and are intended solely for the use of the individual to whom it is addressed.
​Any views or opinions expressed are solely those of the author and do not necessarily represent those of Beech Hall School or those of Chatsworth Schools. 
​If you are not the intended recipient of this email you must neither take any action based upon its contents nor copy or show it to anyone.
​Please contact the sender if you believe you have received this email in error.

florianfesti, 2024-03-04:
Dimensions should be fine but I added the reference for a reason. Check if it still has the 100mm. This is without the thickness of the lines - center line to center line.

BHSDT, 2024-03-05:
Thank you for all your help on this - we really appreciate it.  We managed
to get an SVG file to convert accurately to DXF using CloudConverter online.

Let us know when you manage to fix the bug on your site as it will be good
not to have to do this extra step each time.

Many thanks for your help and expertise in creating the site

Colin

MrCRichardsBSc, Cert.Ed.
Director of Studies, Head of Design & Technology
01625 422192
Beech Hall School is registered with the Department for Education as an authorised independent school. Registration number: 895/6008
Beech Hall School is part of Chatsworth Schools Ltd. Company Number 09540576 
Registered Office: Great Tew Estate Chipping Norton Oxfordshire OX7 4AH
This email and any attachments to it may be confidential and are intended solely for the use of the individual to whom it is addressed.
​Any views or opinions expressed are solely those of the author and do not necessarily represent those of Beech Hall School or those of Chatsworth Schools. 
​If you are not the intended recipient of this email you must neither take any action based upon its contents nor copy or show it to anyone.
​Please contact the sender if you believe you have received this email in error.

AndyLin8090, 2024-03-15:
> There's https://florianfesti.github.io/boxes/html/install.html but I am not sure if it is that helpful for a Newbie.

> Hi Florian

 I tried installing it locally and it opened successfully. However, the local page cannot be saved as a file in DXF format. It only provides svg, svg_Ponoko, ps, lbrn2. No DXF

BHSDT, 2024-03-15:
Thank you.

Cloud Converter has been successful in converting svg to dxf, but we are
limited to 10 free uses per user.

MrCRichardsBSc, Cert.Ed.
Director of Studies, Head of Design & Technology
01625 422192
Beech Hall School is registered with the Department for Education as an authorised independent school. Registration number: 895/6008
Beech Hall School is part of Chatsworth Schools Ltd. Company Number 09540576 
Registered Office: Great Tew Estate Chipping Norton Oxfordshire OX7 4AH
This email and any attachments to it may be confidential and are intended solely for the use of the individual to whom it is addressed.
​Any views or opinions expressed are solely those of the author and do not necessarily represent those of Beech Hall School or those of Chatsworth Schools. 
​If you are not the intended recipient of this email you must neither take any action based upon its contents nor copy or show it to anyone.
​Please contact the sender if you believe you have received this email in error.

AndyLin8090, 2024-03-27:
If you have the software lightburn installed on your computer, you can save the box file you generated as an lbrn2 type file in BOXES.PY, then open it with lightburn and save it as a dxf file, so that you can easily use it in AUTOCAD Edit it here. It can exceed the 10 usage limit of Cloud Converter. Thanks

BHSDT, 2024-03-27:
That's a great hint - I am considering upgrading to lightburn next academic
year, but will have to upgrade the controller hardware on our cutter to do
so

MrCRichardsBSc, Cert.Ed.
Director of Studies, Head of Design & Technology
01625 422192
Beech Hall School is registered with the Department for Education as an authorised independent school. Registration number: 895/6008
Beech Hall School is part of Chatsworth Schools Ltd. Company Number 09540576 
Registered Office: Great Tew Estate Chipping Norton Oxfordshire OX7 4AH
This email and any attachments to it may be confidential and are intended solely for the use of the individual to whom it is addressed.
​Any views or opinions expressed are solely those of the author and do not necessarily represent those of Beech Hall School or those of Chatsworth Schools. 
​If you are not the intended recipient of this email you must neither take any action based upon its contents nor copy or show it to anyone.
​Please contact the sender if you believe you have received this email in error.

ochangus, 2024-03-28:
> Easily reproduced with the web instance. It works here locally with a newer version of ps2edit. Guess I need to update the container to a newer release of the distribution. If you are in a hurry you can try to run Boxes.py locally for now.

Hi Florian, thanks so much for supplying this tool it realy is a massive help with our school kids too. The dxf output not working has been a headache - thye've had to draw them themselves - it's not gone well!! I will ask our tech tream about a local install., Angus

florianfesti, 2024-03-28:
Oh dear! I'll try to get it back up working over the holidays. Til then youcan just create PS files and give them to ps2edit or SVG files and convert them with Inscape.

ochangus, 2024-03-28:
Florian, Thankyou for replying, I'm sorry to cause you work but your
website is a great resource. It will be great to be able to use it fully
again.

Kindest regards, Angus Tulloch

florianfesti, 2024-03-29:
OK, a3f5aa1a48f8b124eae1426f2ea826c125cc9fd7 should fix this for now. Please test and re-open this ticket if there are any issues. Sorry for the inconvenience!

closed 2024-03-29 by florianfesti

BHSDT, 2024-04-11:
Thank you so much - appears to be fully functional again.

This is so much appreciated in our work with pupils.

Kind regards

Colin

MrCRichardsBSc, Cert.Ed.
Director of Studies, Head of Design & Technology
01625 422192
Beech Hall School is registered with the Department for Education as an authorised independent school. Registration number: 895/6008
Beech Hall School is part of Chatsworth Schools Ltd. Company Number 09540576 
Registered Office: Great Tew Estate Chipping Norton Oxfordshire OX7 4AH
This email and any attachments to it may be confidential and are intended solely for the use of the individual to whom it is addressed.
​Any views or opinions expressed are solely those of the author and do not necessarily represent those of Beech Hall School or those of Chatsworth Schools. 
​If you are not the intended recipient of this email you must neither take any action based upon its contents nor copy or show it to anyone.
​Please contact the sender if you believe you have received this email in error.