florianfesti/boxes #641: plt file generator not working

lazarocrafing, 2024-03-03:
By suggesting a new box generator or other feature you agree to test cut it until it works. You understand that this may take a few attempts. As soon as it does work you will provide a sample image that can be used in the user interface or the documentation.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you think would be the best solution.

** Pictures of something similar **
If possible provide a picture of a similar item or feature. If copy right allows it uploading it to this ticket is preferred as it is less likely to disappear.

florianfesti, 2024-03-03:
Duplicate of #637 
Issue is that older versions of pstoedit do not allow stdin as a source file and the web instance works on old crap that is to be replaced.

closed 2024-03-03 by florianfesti

lazarocrafing, 2024-03-03:
I am using boxes.py direct from web it's not generating plt file error code
comes instade

lazarocrafing, 2024-03-03:
this is error on site