florianfesti/boxes #646: Webpages appears to have lost all layout & styling

Karrion14041978, 2024-03-10:
Having some issues with the webpage today.   seems to have lost the majority of the styling and layout.

I have downloaded and installed the latest version to no avail.

Has anyone else had this issue?

[image]

Karrion14041978, 2024-03-10:
Updated Phython and all associated plugins 
Sorry I forgot to mention, this is a Windows install and I have been using it for months without issue

littleroblet, 2024-03-10:
@Karrion14041978 I updated my branch with the latest code and ran the server without any issues. I am also using Windows. 

I had an issue with a project last night doing weird things and spent ages troubleshooting, It turns out it was a cache issue and a hard refresh with ctrl f5 fixed it. Hopefully, you have already tried this and are not being silly like me.

Karrion14041978, 2024-03-11:
Hi @littleroblet, thanks for responding..   Yeah I have tried a hard reset, reboot & multiple browsers with no luck.   In the end I span up a Linux VM and could access this without any issues

So long as this issue is only affecting me you can safely ignore it.   I just thought it may have been a wider issue cause by a windows update or something similar as it came out of the blue.

Thanks again

K

florianfesti, 2024-03-11:
You can try looking int the page sources and follow the link to the `self.css`. May the that way you can see an error on the command line output.

There really were not any changes in that area of code lately. So this is most likely an issue with your setup.

florianfesti, 2024-03-23:
I fear there's not much we can do here as it is most likely a local issue. Closing.

Feel free to still ask question here or re-open if the issue clears up.

closed 2024-03-23 by florianfesti

florianfesti, 2024-03-30:
May be you did use the URL from the web instance and replaced festi.info by localhost:8000 without purging the boxes.py/ from the URL?