florianfesti/boxes #656: Shelf for laptop and phone charger pad

SnowBeaver, 2024-03-30:
By suggesting a new box generator or other feature you agree to test cut it until it works. You understand that this may take a few attempts. As soon as it does work you will provide a sample image that can be used in the user interface or the documentation.

**Describe the solution you'd like**
It is an extension of the Display Shelf with one shelf, a shelf angle at -60 degrees and no shelf lip. However The generated image cannot be used as is and must be edited. The shelf length is too long. There are too many shelf fingers. The shelf should be raised to leave room for the botton charger. 

** Pictures of something similar **
I do not have a picture, I have included the DisplayShelf setting to produce what I think is as close as I want.
[image]

SnowBeaver, 2024-04-01:
Updated description.

florianfesti, 2024-04-02:
Not quite sure why you don't just use a normal DisplayShelf without a back wall. That should be strong enough and with setting `front_wall_height` large enough Things should not fall through. Also you get holes at the back for free.

SnowBeaver, 2024-04-02:
I need the back panel to fix the box to a wall.  That is why I need holes
on the side panels.

Thanks for your reply.

SnowBeaver, 2024-04-04:
Here is the shelf fixed to a small cabinet. Modification that could be added are holes for the usb cables,  support for holding the usb power supply. Thanks
[image]
[image]
[image]

florianfesti, 2024-04-09:
OK, you can now have walls at the bottom the front or the back. Holes for cables need to be added by yourself e.g. using Inkscape.

Pleas have a look if everything fits together.

SnowBeaver, 2024-04-09:
Thanks Florian.

I looked at https://www.festi.info/boxes.py/DisplayShelf?language=en but
the options are the same. I suppose to have a look at your changes I must
run the code using python, but I do not know how to do it, at least I did
not find it.

Michel

florianfesti, 2024-04-09:
Ops, forgot to update the web instance...

Should be there now.

SnowBeaver, 2024-04-10:
I tried it and I have a  few comments:

The bottom need more height with the shelf. it should be variable to adjust to every user needs. 
[image]

With the following settings, the shelf go outside the box and should be clipped

[image]

[image]

florianfesti, 2024-08-03:
9972d329b2d15c91a1e6d6f594fcc4894b9e7f34 and 769236b5df3de9c009b74eda59a57c62e617ed96 fixes placement issues of the shelves and issues an error to the user if the shelves won't fit in the height.

commit 4829dcaf (2024-04-09): DisplayShelf: Add include_front and include_bottom options

This allows using the shelf backwards and using the space under the bottom shelf

Resolves: #656
commit 86bb532f (2024-08-03): DisplayShelf: Add bottom_distance and top_distance

Allow adding more space above and below the shelves.

Resolves: #656
closed 2024-08-03 by florianfesti