florianfesti/boxes #667: Windows build fails with recent master

vmario89, 2024-04-01:
Hi,

the recent master upstream does not install on Windows command line. When running

```
pip install git+https://github.com/florianfesti/boxes.git
```

the following output is generated

```
Building wheels for collected packages: boxes
  Building wheel for boxes (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [24 lines of output]
      running bdist_wheel
      running build
      running build_py
      updatePOT()
      usage: xgettext [OPTION] [INPUTFILE]...
      xgettext: error: unrecognized arguments: -L -j boxes/*.py scripts/boxesserver scripts/boxes
      generate_mo_files()
      Traceback (most recent call last):
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
          return _run_code(code, main_globals, None,
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
          exec(code, run_globals)
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\Scripts\msgfmt.exe\__main__.py", line 7, in <module>
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\lib\site-packages\gettex\msgfmt.py", line 144, in run
          main(
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\lib\site-packages\gettex\sty_renderer.py", line 277, in main
          output_file.write_text(output)
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\lib\pathlib.py", line 1155, in write_text
          return f.write(data)
        File "C:\Users\gurke\AppData\Local\Programs\Python\Python310\lib\encodings\cp1252.py", line 19, in encode
          return codecs.charmap_encode(input,self.errors,encoding_table)[0]
      UnicodeEncodeError: 'charmap' codec can't encode characters in position 1114-1117: character maps to <undefined>
      buildInkscapeExt()
      error: [WinError 2] Das System kann die angegebene Datei nicht finden
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for boxes
  Running setup.py clean for boxes
Failed to build boxes
```

the issue seems to be in https://github.com/florianfesti/boxes/blob/494e445adab71c86da5dfbb8a8dad081a3f51b53/setup.py#L20 but i do not fully understand that line of code at the moment

previously this thing worked

florianfesti, 2024-04-01:
The xgettext and msgfmt shipped with your Python version are kinda weird. They seem to be implemented in Python, while they are binary programs here on Linux.

While https://github.com/florianfesti/boxes/blob/494e445adab71c86da5dfbb8a8dad081a3f51b53/setup.py#L20 

creates warnings about not supported parameters the actual error comes from

https://github.com/florianfesti/boxes/blob/494e445adab71c86da5dfbb8a8dad081a3f51b53/setup.py#L31

May be you are missing a native `gettext` implementation that was used previously. Or it is no longer found earlier in the path.

vmario89, 2024-04-01:
Hi Florian. Thank you for looking into it.

Indeed i just missed to install https://github.com/mlocati/gettext-iconv-windows/releases/download/v0.21-v1.16/gettext0.21-iconv1.16-static-64.exe. but it continues with another issue now:

```
Building wheels for collected packages: boxes
  Building wheel for boxes (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      running bdist_wheel
      running build
      running build_py
      updatePOT()
      xgettext: Öffnen der Datei »boxes/*.py« zum Lesen fehlgeschlagen: Invalid argument
      generate_mo_files()
      buildInkscapeExt()
      error: [WinError 2] Das System kann die angegebene Datei nicht finden
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for boxes
  Running setup.py clean for boxes
Failed to build boxes
ERROR: Could not build wheels for boxes, which is required to install pyproject.toml-based projects
```

i checked my really old legacy documentation but it seems to be an issue which already exists for years, as i had the same problem in 2019 or so.

The following procedure will make boxes.py work or install and therefore run on cli or inkscape, but we have to do a lot of manual steps. What works:

```
git clone https://github.com/florianfesti/boxes.git boxes

cd boxes
 
#fix PO/MO translations manually by running the following commands:
xgettext -L Python -j --from-code=UTF-8 -o po/boxes.py.pot C:\boxes\po\boxes.py.pot -D C:\boxes\boxes
xgettext -L Python -j --from-code=UTF-8 -o po/boxes.py.pot C:\boxes\po\boxes.py.pot -D C:\boxes\boxes\generators
xgettext -L Python -j --from-code=UTF-8 -o po/boxes.py.pot C:\boxes\po\boxes.py.pot -D C:\boxes\scripts\boxesserver
xgettext -L Python -j --from-code=UTF-8 -o po/boxes.py.pot C:\boxes\po\boxes.py.pot -D C:\boxes\scripts\boxes
  
#returns errors like above - ignore that
python.exe setup.py build
 
#returns errors - ignore
python.exe setup.py install
```

 The best for regular users will be to install boxes.py directly with `pip install git+<mirror>` and thats it.

btw. on the website documentation the qrcode lib is not mentioned -> https://florianfesti.github.io/boxes/html/install.html

vmario89, 2024-04-01:
`xgettext: Öffnen der Datei »boxes/*.py« zum Lesen fehlgeschlagen: Invalid argument` get definetely produced by the wildcard character. i am successfull running `xgettext -L Python -j --from-code=utf-8 -o po/boxes.py.pot boxes/*.py scripts/boxesserver scripts/boxes` on Windows cmd shell manually. I already tried to switch or escape that stuff. Maybe xgettext on Windows is not capable handling this ...

vmario89, 2024-04-01:
the following seems to work on Widows:

```
dir /s /b /o:gn boxes\*.py > PYFILES
xgettext -L Python -j --from-code=utf-8 -o po/boxes.py.pot -n --files-from=PYFILES scripts/boxesserver scripts/boxes
```

vmario89, 2024-04-01:
The `error: [WinError 2] Das System kann die angegebene Datei nicht finden` comes from inkscape not being on %PATH%

florianfesti, 2024-04-13:
Ok, fixed this slightly differently. The try except block was supposed to already catch this kind of failure but didn't.

I hope this allows you to get this finally built. Thanks for the thorough analysis!

commit b1bd75e5 (2024-04-13): setup.py: Handel missing Inkscape properly

Resolves: #667
closed 2024-04-13 by florianfesti