florianfesti/boxes #670: svg from any font

PetitPhoenix, 2024-04-10:
Hello, I'm a newbie in python but initiated the coding of a module that creates svg for any ttf font. Your repository and the website you developped inspired me in going that route. Now that I have a code somehow running, I might start the website part :)

So I was wondering if you were interested in the feature of creation of svg from any font. You may have it already available and I would like to know how you handled it.

If interested, I'd like to know if you had any good practices so that the contribution I would make would fit best in your framework.

Sorry if the "Issue" is not the right way to ask, newbie in github, I didn't know how to contact. Thanks.

florianfesti, 2024-04-13:
Right now I am not using any font libraries. Boxes.py generates text. So it leaves the rendering of the fonts to the client on all supported formats. This has pros and cons.

The things is if you are only interested in generating texts in different fonts you, too, can leave that to the client side. SVG has much more powerful mechanisms to alter the text that I am using in Boxes.py (which tries to only use simple graphic primitives that can be turned in to different formats).

So unless you want to do something very specific it might not e worth it to reinvent the wheel and create your own ttf library. Especially as there are probably already a few out there. May be you can find one that can be used for the ttf part so you only need to turn the paths into an SVG format, whihc should be relatively simple.

KittyDreadful, 2024-04-23:
I’d like to know more about converting TTF to SVG fonts.   This probably isn’t the right place.  Can I contact you separately?

florianfesti, 2024-04-24:
> I’d like to know more about converting TTF to SVG fonts. This probably isn’t the right place. Can I contact you separately?

Nah, this is pretty much on topic here.

The issue is more that I do not know much about that topic. In the past Boxes.py was using libcairo as a backend. libcairo puts text in as a graphics as some of its output formats don't have text but are true graphics.

Here is an example on the produced SVG: view-source:https://raw.githubusercontent.com/florianfesti/boxes/6ab0cef413c110947d20b5fd3d414d170c7a2a13/examples/drillbox.svg

So each letter is declared as some
```
<symbol overflow="visible" id="glyph0-0">
```
containing a path that draws the letter.

And then used later with a

```
<use xlink:href="#glyph0-4" x="105.101562" y="333.414062"/>
```

If that is good enough for you you can use the SVG backend of libcairo. That way you do not need to bother with the details of the fonts at all. You just set the font and place some text sopmewhere and the library handles turning the ttf glyph into a SVG symbol. For Boxes.py we ofc don't want to go back to using libcairo.

PetitPhoenix, 2024-04-25:
Hello, on my side, I used fonttools to get the glyphs, kerning and spacings and convert to path. The code I created try to get the right scale to fit in the area you want it to fit. The need was to try to automate create of small tags (in stl) and wifi qr codes.
Indeed it has pros and cons. Basic thing was that when I create a svg that I use on another computer, using a ttf font that was not on that computer was an issue => did not render as expected. I guess it's the same when you deliver it to someone.
I tried to use libcairo but on my computer again, didn't work (don't know why actually)

Stéphane

florianfesti, 2024-04-26:
Yeah, libcairo is a pain - especially if you are not on Linux.

Also a quick Google search suggest that you might be able to embed the fonts in the SVG. Either via link are actually embedding them base64 encoded. That's probably the best way if you just want text to render correctly.

florianfesti, 2024-07-06:
Ok, guess we need a decision here:I don't really think a simply text to graphics generators really fits in to boxes.py. One can also just use Inkscape and convert some text to a path. There many other operators are also available and a simple web site won't be able to compete with that. As far as Boxes.py itself is concerned I think just adding text is fine and making graphics out of it does not really add much functionality. Also Boxes.py would need to ship all the fonts locally or support uploading them. This is all much more convenient with a local desktop program.

So while your TTF to SVG project is interesting I don't think it fits into Boxes.py at the moment.

closed 2024-07-06 by florianfesti