florianfesti/boxes #674: Failing test: Shadowbox sample

gcollic, 2024-05-11:
**Describe the bug**

When running the commands from the contributing guidelines, a test is failing.

**To Reproduce**

When running `pre-commit run --all-files` or directly `pytest`, the test `TestSVG.test_generator[Shadowbox]` is failing (`SVG files are not equal. If change is intended, please update example files.`).

Indeed, after executing `python .\scripts\boxes --examples`, `git status` shows `modified: examples/Shadowbox.svg`.

I suppose updating the example is needed, but not knowing this box I'm not sure if the difference is right or wrong.

**Screenshots or Drawings**

[image]

**Additional context**

The added 4 shapes are surely from #672, but I'm not sure about the other 2 disappearing.

florianfesti, 2024-05-11:
I somehow count 14 pieces for both sides...

Looks light they just got a bit thinner.

florianfesti, 2024-05-11:
d2422e43256bcd22c2b49e80e0f54becfc2096bb fixes this en passant.

closed 2024-05-11 by florianfesti

gcollic, 2024-05-11:
Thanks!
And sorry for the misdirection, the view was scaled differently in inkscape, I was mistaken