florianfesti/boxes #677: UniversalBox.py Default Settings

ScottCTU, 2024-05-17:
UniversalBox Under default settings "inner_corners" drop down menu changes inner corner style for inner corner to CORNER  LOOP;  CORNER; BACKARC.
This does not change the ARC on the finger joint

Like to add similar default setting style for outer corner "outside_corners" ARC; CORNER.

[image]

florianfesti, 2024-05-17:
See https://florianfesti.github.io/boxes/html/api_burn.html for the details on the burn correction.

I don't really see a benefit of making the outer corners sharp. Do you have a particular use case or a process that would benefit from this change?

Note that you can always generate the drawing with zero burn and do the burn correction elsewhere - like in the CAM software or the control software of you laser cutter / CNC machine.

ScottCTU, 2024-05-17:
I use boxes.py universal box with new export feature in the format lbrn2 (lightburn)
LightBurn Has the ability  to resize the slot depth, Change material thickness.

[cid:90a39e83-c471-4c01-91c4-c04c2a3b71b9]

With Universal Box under Default settings inner_corners i can select corner (90 degree) not loop and the lightburn Resize Slots partially works.
The problem in light burn the tab height is reduced by the line being shorter with the arc from boxes.py
If another setting in Universal boxes for outer_corners could square the corners, then no arc.

[cid:95ffedc2-dd15-49d4-82d8-623324258fe3]

In addition LightBurn has a radius function, simply select degree and click on the nodes that can add/change/delete  raduis  from a 90 corner. Doesn't work with the boxes.py lbrn2 imported file

florianfesti, 2024-07-31:
Changing the outer corners is too much of an interference on how the kerf correction is working. Please try a burn value of zero and go from there. I don't plan to support this half here half there work flow. Sorry for the inconvenience.

closed 2024-07-31 by florianfesti