florianfesti/boxes #679: Picture/photo frame generator

marauder37, 2024-05-26:
I would like to generate picture/photo frames similar to [this YouTube tutorial](https://www.youtube.com/watch?v=oXql_iFCe4Q&t=303s)

[image]
[image]

It is a 3-layer design meant to be cut from 3mm plywood/MDF. Basically it makes a pocket that a piece of clear acrylic slides into. The layers are:

* back layer that the picture will be glued to. A rectangle with the full dimensions of the final frame, with an etched rectangle marking where to glue the picture, and a mounting hole hidden by the top layer
* middle layer to hold a piece of acrylic. 3 strips joined with dovetails.
* top layer. In the video, this is another rectangle with the full dimensions of the final frame, and a window cut out that is smaller than the piece of acrylic.

Some options that would make the design in the video more flexible/useful:

* assemble the top layer from 4 strips joined with dovetails, instead of cutting from 1 rectangle, to minimise material use
* generate a matting part to be cut from cardboard/matboard instead of the acrylic sheet 

I would try to make this generator myself, but I can't find any examples of joining parts with dovetails, so I'm not sure where the best place to start is.

florianfesti, 2024-05-26:
Finger joints are just another type of edge. They are available as `d` and `D` edges. See https://github.com/florianfesti/boxes/blob/master/boxes/generators/shadowbox.py as an example that is already pretty close to what you want. Although https://github.com/florianfesti/boxes/blob/master/boxes/generators/airpurifier.py has pieces that are closer to those above.

These pages of the API docs might also be helpful:

* https://florianfesti.github.io/boxes/html/api_parts.html#the-edges-parameter
* https://florianfesti.github.io/boxes/html/api_existing_parts.html
* https://florianfesti.github.io/boxes/html/api_edges.html

florianfesti, 2024-05-26:
I am totally not taking any responsibility for any air purifiers you build.

marauder37, 2024-06-30:
p.s. I did build an air purifier, because an air filter just showed up out of nowhere and it would be a shame to waste it. But I heroically resisted all impulses to over-engineer it or do a good job.

[image]

florianfesti, 2024-07-16:
That's the right spirit!

With #680 merged I guess we can close this one.

closed 2024-07-16 by florianfesti